talkkonnect/talkkonnect #4: Vox option

tomjones1977, 2019-02-01:
If using talkkonnect to bridge a radio as an RF gateway is there a way to enable vox transmission?  I want the tx to be keyed by the radio receiving an RF transmission.

closed 2019-02-05 by talkkonnect

talkkonnect, 2019-02-05:
There is some python code that one of our community members has used before with talkkonnect successfully,

Chris-The-Tuner, 2019-07-11:
I tested it and i must Say the Vox works (kind of) since only 1 program is allowed to use the Audio Source.

If i start the Vox script when talkkonnect is running i get a:
alsaaudio.ALSAAudioError: Device or resource busy [default]

If i start the Vox BEFORE talkkonnect...
Then there is NO AUDIO in Talkkonnect :(

zrdimetc, 2019-07-31:
Chris, no issues with pyalsaaudio vox here. Script works regardless if started before or after talkkonnect launches. It works to detect audio / voice and set the PTT GPIO pin high. But preferably start the script after talkkonnect launches to give enough time for a connection to Mumble server to be established and talkkonnect to "settle" before PTT button is pressed. Try creating local alsa config .asoundrc ... Are there any other programs accessing Alsa sound system? How do you start / stop talkkonnect? If it crashed, did you kill old instances of program? Are you detecting voice from a condenser microphone or are you feeding some other audio signal to the sound card mic input? Is mic working when you try to record voice or with other sound programs? Do you use Raspberry Pi or some other board? Trying to understand why your vox isn't working.

Chris-The-Tuner, 2019-07-31:
I have trouble gettin it to run here.
I'm using a Raspi 3

I did kill all fails (rebooted)

I have the local .asoundrc in place.

There is nothing else running so nothing is using alsa, only talkkonnect and the vox

I'm feeding Audio from a CB Radio via a small filter into the Sound card (cheap generic china stuff)

I will try to start from scratch again and see if i mayby made a mistake.