talkkonnect/talkkonnect #6: select audio-interface

yasdfgr, 2019-05-02:
Is there a possibility to select a audio-interface e.g. to say explicit "use that interface" in talkkonnect / gumble?
I want use two instances of talkkonnect on one raspberry pi.

talkkonnect, 2019-05-03:
I have not tried that but I guess you can set up 2 sound cards and then run talkkonnect with separate config file for each instance and the xmltag for example <outputdevice>Speaker0</outputdevice> and <outputdevice>Speaker1</outputdevice>. You will have to name these devices in alsa config. And make sure that you dont define a tx pin on one of the two config files otherwise you will be transmitting on both instances.

let me know how you get along.

yasdfgr, 2019-05-04:
Can you describe the "todo" a little more in detail? I'm wondering how it could be possible because in the README is a description about changing the default sound interface. So i assume that talkkonnect always use the default card. Is there any parameter in the *.xml in addition to the interface name for the mute function?

I read also a little bit thru the source code and couldn't find any hint about which sound interface is used by the gumble client.

Thanks in advance!

talkkonnect, 2019-05-04:
for me when I run 2 instances of talkkonnect I use the same sound card. The reason for running 2 in my case is to monitor 2 channels at one time. I guess with some help from docker or some sort of kvm you can run 2 instances but this has not been tried before.

talkkonnect, 2019-05-09:
allow me to close this ticket if there are no further developments

thanks for the feedback and the interest in using talkkonnect

closed 2019-05-09 by talkkonnect