talkkonnect/talkkonnect #7: Open Mic on startup

GTiDon, 2019-06-15:
Hello
I need my client devices to have open mics on startup. If I have PTT latched at startup it does not work. Is there something I can change to make this possible?

zrdimetc, 2019-06-16:
Hi, Talkkonnect has a radio-like PTT logic. When one client is talking, it is muting its own sound output to prevent audio feedback between speaker and microphone. It starts in listening mode. But if you want talkkonnect to automatically start transmitting immediately upon a connection to server, you can probably add that function in client.go somewhere after the server connection part? Look at how TX function / F8 is implemented in client.go? Microphone must be “captured” in order for audio input to work. Is your sound system configured correctly? Is Alsa audio input/ output working with other programs? PTT can work as normal momentary push button or be configured as "toggle" key. Keyboard shortcut for PTT is F8. Release PTT with F9. Do you use GPIO for PTT button? In talkkonnect xml config you can specify "PC" or "rpi" mode. In PC mode GPIO functions are disabled. Please give more details about your build and what exactly is not working with PTT or mic, how it fails and I will try to help? I built many talkkonnects, they all work "as advertised".

zrdimetc, 2019-06-16:
Hi, Talkkonnect has a radio-like PTT logic. When one client is talking, it is muting its own sound output to prevent audio feedback between speaker and microphone. It starts in listening mode. But if you want talkkonnect to automatically start transmitting immediately upon a connection to server, you can probably add that function in client.go somewhere after the server connection part? Look at how TX function / F8 is implemented in client.go? Microphone must be “captured” in order for audio input to work. Is your sound system configured correctly? Is Alsa audio input/ output working with other programs? PTT can work as normal momentary push button or be configured as "toggle" key. Keyboard shortcut for PTT is F8. Release PTT with F9. Do you use GPIO for PTT button? In talkkonnect xml config you can specify "PC" or "rpi" mode. In PC mode GPIO functions are disabled. Please give more details about your build and what exactly is not working with PTT or mic, how it fails and I will try to help? I built many talkkonnects following Suvir's excellent instructions, they all work "as advertised".

zrdimetc, 2019-06-16:
Hi, Talkkonnect has a radio-like PTT logic. When one client is talking, it is muting its own sound output to prevent audio feedback between speaker and microphone. It starts in listening mode. But if you want talkkonnect to automatically start transmitting immediately upon a connection to Mumble server, you can maybe call tx function TransmitStart() in client.go somewhere after the server connection part? Look at how TX function / F8 is implemented in client.go? Or you could try change PTT button behavior in gpio.go? Microphone must be “captured” in order for audio input to work with talkkonnect.

talkkonnect, 2019-06-24:
Could you share what it is you are trying to do? What is the use case? How will you stop it from transmitting?

closed 2019-07-30 by talkkonnect