talkkonnect/talkkonnect #8: RaspiAudio MIC+

snarkulon, 2019-08-13:
Has anyone gotten talkkonnect to work with a raspiaudio MIC+ board.  It has built in speakers, a mic, a button and an LED and should make a nice little voip intercom but I can't get talkkonnect to work with it.

talkkonnect, 2019-08-27:
I have not tried it personally but heard many people have problems with the raspi speaker as it is not detected the same as a normal USB sound card.

qabo0101, 2020-01-01:
Having som issues:

talkkonnect@raspberrypi:~/gocode $ go get github.com/talkkonnect/talkkonnect
package math/bits: unrecognized import path "math/bits" (import path does not begin with hostname)

talkkonnect@raspberrypi:~/gocode/src/github.com/talkkonnect/talkkonnect $ go build -o /home/talkkonnect/bin/talkkonnect cmd/talkkonnect/main.go
../../../golang.org/x/sys/unix/affinity_linux.go:10:2: cannot find package "math/bits" in any of:
        /usr/lib/go-1.7/src/math/bits (from $GOROOT)
        /home/talkkonnect/gocode/src/math/bits (from $GOPATH)

Thank you / qa

closed 2020-02-25 by talkkonnect