talkkonnect/talkkonnect #9: trying to get oled to work

davidywilson, 2019-10-27:
Hi 
Im trying to get OLED to work but it seems that the program is expecting additional config in the xml that is not in the template. 
for eg: if i run talkkonnect and press ctl-x i get 

2019/10/27 17:33:50 info: ------------ OLED ----------------------- 
2019/10/27 17:33:50 info: Enabled                 false
2019/10/27 17:33:50 info: Interfacetype           
2019/10/27 17:33:50 info: DisplayRows             0
2019/10/27 17:33:50 info: DisplayColumns          0
2019/10/27 17:33:50 info: DefaultI2cBus           0
2019/10/27 17:33:50 info: DefaultI2cAddress       0
2019/10/27 17:33:50 info: ScreenWidth             0
2019/10/27 17:33:50 info: ScreenHeight            0
2019/10/27 17:33:50 info: CommandColumnAddressing 0
2019/10/27 17:33:50 info: AddressBasePageStart    0
2019/10/27 17:33:50 info: CharLength              0
2019/10/27 17:33:50 info: StartColumn             0

that is a separate section to the lcd section - so is there new config settings that need to be used? you also state in the readme that there is an oled section and the need to set display height and width. 

i assume there is additional information such as enabling oled in raspi-config, and also setting permissions to i2c 

can you help me work oput what is required to make the oled work?

PS. nice work with all this - It is nice to find quality programming these days.

zrdimetc, 2019-11-03:
Let me try to help. In the `sample-configs` folder you will find xml configs for different platforms (Raspberry Pi, Orange Pi, PC) and OLED i2c or HD44780 parallel displays. Use the correct template for your build and replace the default xml config from talkkonnect folder, which is for HD44780 display. Currently, only i2c Oled is supported in Talkkonnect. Not yet spi. 1.3 inch or 0.96 inch Oled's can be used. If you use 0.96, adjust "`oledscreenwidth`" to 128 (130 for 1.3) and "`oledstartcolumn`" can be "0" (instead of col 1 for 1.3 "). But this is only to help you align the screen text for a particular Oled geometry. For Oled to initialize you must match the i2c bus number and display address in the xml config with actual i2c bus you use and the display address. Read the hex address usually printed at the back of Oled and convert that to decimal number. Use that dec number in the xml config (`oleddefaulti2caddress`). You can also install `i2c-tools` package and use command `i2cdetect -y <i2c bus number>` ... 1 or 0 for bus number to verify Oled address, is it connected correctly and communicating? Of course you need to enable i2c bus in `raspi-config` or use whatever other method. Why dont you try your Oled display first with some more simple third party library and example program? Choose any Oled libraries or examples from github with the aim to try to initialize your Oled display and write some text to screen? After verifying Oled settings and running some simple tests or examples to build confidence about your display, go back to Talkkonnect and try Oled working again? No need to change i2c permissions if you follow install recommendations. Oled's will work great with Talkkonnect, the only thing that can go wrong is if you accidentally connect them to a wrong polarity. Some Oled pins start with Gnd, other with Vcc, so never fully trust the manufacturer diagrams without inspecting the actual display. You can run them at 5V or 3.3V. 3.3V should be a little safer. Voltage regulator in Oled's will tolerate up to 6V, that is a very small margin from 5. If you need any more help, just let us know.

davidywilson, 2019-11-04:
Thanks 

I did find the sample configs and set it up. 

I only have a PiOLED small display 
It is 128x32 and works with other software 

It is now trying to display - but its not right 
It seems to not ave enough pixels for the height of the text.

I know you said 

> 	Currently, only i2c Oled is supported in Talkkonnect. Not yet spi. 1.3 inch or 0.96 inch Oled's can be used. If you use 0.96, 

But is there a way to adjust the setting for the half height screen? 
You have “oleddisplayrows” setting - I would have assumed that setting should allow me to reduce the number of lines of text that you attempt to display. 

Its not a huge thing as I can get a 1.3” screen - but this one is all I have here right now. 

Regards 
Dave Wilson

zrdimetc, 2019-11-04:
0.91" 128x32 Oled's are usually very similar to 0.96" 128x64 and probably use the same SSD1306 chip. Initialization should work the same. Try to tweak `client.go` to change all instances where Oled function is called? It will look like this `oledDisplay(false, row_number, column_number, "Some text")`. Rearrange writing to Oled, so that you never exceed 4 rows (0-3). Missing 4 rows, there will be some info that you wont be able to show. Just decide what info to comment out and what to keep to stay within rows 0-3? It might work. If more issues, have a look at `oleddisplay.go` and `go-oled-i2c` supporting library if anything else needs to be adjusted there as well? But try tweaking `client.go` first, let us know? 0.96" is probably the most widely used Oled in projects, followed by 1.3". 0.91" is not that uncommon. I am not sure if Suvir is willing to add support for 0.91"? Supporting 0.96" and 1.3" was logical step. They are low cost, easily available, similar geometry, it was just a minor config change to support them both without engaging in too much code redesign. 1.3" is easy to read. 0.96/0.91 can cause some strain.
Regards, Zoran

zrdimetc, 2019-11-04:
0.91" 128x32 Oled's are usually very similar to 0.96" 128x64 and probably use the same SSD1306 chip. Display initialization should work the same. 

- In `talkkonnect.xml` redefine Oled height and width for 0.91" to:

`<oledscreenwidth>128</oledscreenwidth>`
`<oledscreenheight>32</oledscreenheight>`

Test. If still having problems,

- Try to tweak client.go to change all instances where Oled function is called? It will look like this 

`oledDisplay(false, row_number, column_number, "Some text")`. 

Rearrange writing to Oled, so that you never exceed 4 rows (0-3). Missing 4 rows, there will be some info that you wont be able to show. Just decide what info to comment out and what to keep to stay within rows 0-3?

- If any more issues, have a look at `oleddisplay.go` and `go-oled-i2c` supporting library if anything else needs to be adjusted there as well? But try defining 128x32 Oled geometry in talkkonnect.xml and /or tweaking `client.go` first, let us know? 

Regards, 
Zoran

closed 2020-02-25 by talkkonnect