talkkonnect/talkkonnect #11: import cycle not allowed

wisentius, 2020-02-05:
Hi,

When I start the build (after following all steps) I got following error:

talkkonnect@raspberrypi:~/gocode/src/github.com/talkkonnect/talkkonnect $ go build -o /home/talkkonnect/bin/talkkonnect cmd/talkkonnect/main.go
import cycle not allowed
package main
        imports flag
        imports errors
        imports internal/reflectlite
        imports runtime
        imports internal/bytealg
        imports internal/cpu
        imports runtime
talkkonnect@raspberrypi:~/gocode/src/github.com/talkkonnect/talkkonnect $ 

I'm not a Go expert and have no idea where to look for...

David

talkkonnect, 2020-02-14:
What version of go are you using? On you command line try
go version
you should use version 1.9 and above.
look here https://gist.github.com/random-robbie/1f7f94beda1221b8125b62abe35f45b6
however change the go to the latest version in this line
wget https://storage.googleapis.com/golang/go1.10.1.linux-armv6l.tar.gz
make sure you get the arm version.

try again and let me know if it works. I will need to update the instructions for this when I have time

talkkonnect, 2020-02-15:
I have updated the README.md file to show how to install golang pre-compiled binaries. Using apt-get to install golang does not give you the golang version needed to compile and run talkkonnect.

closed 2020-02-25 by talkkonnect