talkkonnect/talkkonnect #13: Raspberry pi zero

stiopa-lichodiejew, 2020-02-20:
Hello,

I'm stuck with a really small enclosure. Raspi, USB audio and batteries need to fit there. I started with B+ but it takes over half of the available space. So I'm thinking about alternatives. Do you think talkkonnect will run on Raspberry pi zero?

Cheers

zrdimetc, 2020-02-22:
Not very well. Orange Pi Zero is better, it has integrated audio and more powerful Allwinner H2 chip. Raspberry Pi Zero doesnt support Neon. You need Openal without Neon. Daniel Chote has a workaround in his Talkiepi project, but this was working on older Raspbian, not sure if it will work anymore on new. Also you need a way to shut down system when batteries are low. CPU usage will be high on Raspberry Pi Zero. OLED displays are small and supported on talKKonnect. If you can squeeze in Orange Pi Zero you will save yourself a lot of trouble, on condition you take some time to familiarize with Orange Pi's and Armbian.

talkkonnect, 2020-02-23:
Whats the use case? Like Zoran said Orange Pi Zero works well. Raspberry Pi Zero does not since some parts of the CPU are cut out. Let us know how you get along.

stiopa-lichodiejew, 2020-02-23:
Thanks for your replies. The case is I want to retrofit Raspi with Talkkonnect in an old Yaesu VX-2000 along with Arduino to control the front panel. The latter I already solved using Arduino Nano. I also installed Talkkonnect on Raspi 3B+ but the board is just to big to fit in the enclosure along with USB audio and batteries. This is why I was thinking about the Zero. For now I'll stick with 3B+ and batteries attached to the enclosure with velcro. But in the end of the day Orange Pi will eventually take it's place because of yet another reason - the external antenna.
I'll share the results with you soon.

Cheers

closed 2020-02-25 by talkkonnect

zrdimetc, 2020-03-07:
You will need a larger than usual heat-sink on Orange Pi Zero, 20x20x15 mm or you can think of some other shorter but still massive enough heat-sink option that can be fitted within your design, for  keeping the OPi Zero cool. RPi doesnt necessarily need pull up resistors, they are internal, but Orange Pi needs external resistors for reliable button operation. [Ben Lewis](https://github.com/CustomMachines) had made [his project work with RPI Zero](https://github.com/CustomMachines/talkiepi). But be aware that Gumble/ Barnard and talKKonnect use 48k audio sampling. This works near perfectly with official Mumble clients and Plumble. As a result CPU use is high. No problem for RPi 3B+, but RPi Zero will be struggling. Daniel Chote when making talkiepi used 16k audio sampling. This made sense for lowering the CPU usage. But there will be some issues. You can change audio sampling rate by tweaking Gumble files and also frame sizes for conserving some bandwidth. This will work for talkkonnect to talkkonnect or talkiepi to talkiepi communication. No issues. But any other sampling than 48k and 10ms frames will not work very well to Plumble and to a smaller extent official Mumble. You will happy choppy audio. Not entirely clear why this is happening. This is another reason why we dont recommend Raspberry Pi Zero. Orange Pi Zero is a minimum requirement. RPi Zero CPU usage will be 96% on TX. OPI Zero CPU usage is 40 % on TX. Unless you compile OpenAL without Neon support yourself, you will be unable to make either talKKonnect or talkiepi run on Debian Buster. If you need to compare more notes about running talKKonnect on Orange Pi Zero, just let us know as you go forward with your project. Many people have built talKKonnect units with OPi Zero. Regards.

zrdimetc, 2020-03-07:
You will need a larger than usual heat-sink on [Orange Pi Zero](http://www.orangepi.org/orangepizero), 20x20x15 mm or you can maybe think of some other shorter but still massive enough heat-sink option that can be fitted within your design, for keeping the OPi Zero cool. RPi doesnt necessarily need pull up resistors, they are internal, but Orange Pi needs external resistors for reliable button operation. [Ben Lewis](https://github.com/CustomMachines) had made [his project work with RPI Zero](https://github.com/CustomMachines/talkiepi). But be aware that [Gumble](https://github.com/layeh/gumble)/ [Barnard](https://github.com/layeh/barnard) and [talKKonnect](https://github.com/talkkonnect) use 48k audio sampling. This works near perfectly with [official Mumble client](https://www.mumble.info) and [Plumble](https://github.com/acomminos/Plumble). As a result CPU use is high. No problem for RPi 3B, 3B+ and above, but RPi Zero will be struggling. Daniel Chote when making [talkiepi](https://github.com/dchote/talkiepi) used 16k audio sampling. This made sense for lowering the CPU usage. But there will be some issues. You can change audio sampling rate by tweaking Gumble files and also frame sizes for conserving some bandwidth. This will work for talKKonnect to talKKonnect or talkiepi to talkiepi communication. No issues. But any other sampling than 48k and 10ms frames will not work very well to Plumble and to a smaller extent official Mumble. You will happy a "choppy" audio. Not entirely clear why this is happening. This is another reason why not to use Raspberry Pi Zero for talKKonnect. Orange Pi Zero is a minimum requirement. RPi Zero CPU usage will be 96% on TX. OPI Zero CPU usage is 40 % on TX. Unless you compile OpenAL without Neon support yourself, you will be unable to make either talKKonnect or talkiepi run on Debian Buster. If you need to compare more notes about running talKKonnect on Orange Pi Zero, just let us know as you go forward with your project? Many people have built talKKonnect units successfully with OPi Zero. Regards.

zrdimetc, 2020-03-07:
You will need a larger than usual heat-sink on [Orange Pi Zero](http://www.orangepi.org/orangepizero), 20x20x15 mm or you can maybe think of some other shorter but still massive enough heat-sink option that can be fitted within your design, for keeping the OPi Zero cool. RPi's do not necessarily need pull up resistors, they are internal, but Orange Pi Zero needs external resistors for reliable button operation. [Ben Lewis](https://github.com/CustomMachines) had made [his project work with RPI Zero](https://github.com/CustomMachines/talkiepi). But be aware that [Gumble](https://github.com/layeh/gumble)/ [Barnard](https://github.com/layeh/barnard) and talKKonnect use 48k audio sampling. This works near perfectly with [official Mumble client](https://www.mumble.info) and [Plumble](https://github.com/acomminos/Plumble). As a result CPU use is high. No problem for RPi 3B, 3B+ and above, but RPi Zero will be struggling. [Daniel Chote](http://projectable.me/tag/talkiepi) when making [talkiepi](https://github.com/dchote/talkiepi) used 16k audio sampling. This made sense for lowering the CPU usage. But there will be some issues. You can change audio sampling rate by tweaking Gumble files and also frame sizes for conserving some bandwidth. This will work for talKKonnect to talKKonnect or talkiepi to talkiepi communication. No issues. But any other sampling than 48k and 10ms frames will not work very well to Plumble and to a smaller extent official Mumble. You will happy a "choppy" audio. Not entirely clear why this is happening. This is another reason why not to use Raspberry Pi Zero for talKKonnect. Orange Pi Zero is a minimum requirement. RPi Zero CPU usage will be 96% on TX. OPI Zero CPU usage is 40 % on TX. Unless you compile OpenAl without Neon support yourself, you will be unable to make either talKKonnect or talkiepi run on RPI Zero and Debian Buster. If you need to compare more notes about running talKKonnect on Orange Pi Zero, just let us know as you go forward with your project? Many people have built talKKonnect units successfully with OPi Zero. Regards.