talkkonnect/talkkonnect #14: Push to talk GPIO button

umekzg, 2020-02-23:
When i press a couple of times the PTT button the application crashes with the following error:

[  info ] 2020/02/23 01:26:48 Tx Button is pressed
[  info ] 2020/02/23 01:26:51 Tx Button is pressed
[  info ] 2020/02/23 01:26:53 Tx Button is pressed
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x4 pc=0x3f3744]

goroutine 63 [running]:
github.com/talkkonnect/go-openal/openal.(*CaptureDevice).CaptureSamples(...)
        /home/talkkonnect/gocode/src/github.com/talkkonnect/go-openal/openal/alcCore.go:197
github.com/talkkonnect/talkkonnect.(*Stream).sourceRoutine(0x25ca0c0)
        /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/stream.go:284 +0x178
created by github.com/talkkonnect/talkkonnect.(*Stream).StartSource
        /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/stream.go:123 +0x84

Rpi Zero, everything else is fine.

umekzg, 2020-02-23:
on rpi3b+ there are no problems so it seems that zero is just to weak...

talkkonnect, 2020-02-24:
For talkkonnect and other CPU intensive functions I would suggest you stay away from the raspberry pi zero. RBP 3+ can be Model A or B, Raspberry 4 works and is production quality.

closed 2020-02-25 by talkkonnect