talkkonnect/talkkonnect #16: client connect TCP or UDP to server

umekzg, 2020-03-04:
Where can i set Force TCP mode?

zrdimetc, 2020-03-07:
Since Suvir‘s [talKKonnect](url) project is a fork of talkiepie from [Danel Chote](https://github.com/dchote/talkiepi), adding an incredible set of new features bringing a talkiepi concept to a new level and both talKKonnect and talkiepi were made possible thanks to excellent groundbreaking innovative work of [Tim Cooper](https://github.com/layeh/gumble) who created Gumble libraries to make Go Mumble client and both talkiepi / talKKonnect possible, all of the 3 popular Go clients ([Barnard](https://github.com/layeh/barnard), talkiepi and talKKonnect) only support TCP. Tim Cooper, has a branch of Gumble that supports UDP, but neither of the clients officially support switching between UDP and TCP or forcing either. All of the Go Mumble clients based on Gumble code use TCP. If Gumble code once start supporting both UDP/ TCP, this will probably be reflected in talkiepi and talKKonnect, at such time. But as far as I understand neither Tim, Daniel or Suvir probably wont be going there unless there is some strong motivation. Perhaps there will be more people interested, who may want to contribute to UDP mode in Gumble and then forcing TCP or UDP would be possible, maybe through an excellent XML configuration that Suvir had created for 
setting talKKonnect. Right now there is nothing to force. Barnard, talkiepi and talKKonnect are all TCP only.

zrdimetc, 2020-03-07:
Since Suvir‘s [talKKonnect](https://talkkonnect.com) project is a fork of [talkiepie](https://github.com/dchote/talkiepi) from [Danel Chote](http://projectable.me), adding an incredible set of new features bringing a talkiepi concept to a new level and both talKKonnect and talkiepi were made possible thanks to excellent groundbreaking work of [Tim Cooper](https://layeh.com) who created [Gumble](https://github.com/layeh/gumble) libraries to make Go Mumble clients and both talkiepi / talKKonnect possible, all of the clients based on Gumble ([Barnard](https://github.com/layeh/barnard), [talkiepi](https://github.com/dchote/talkiepi) and [talKKonnect](https://github.com/talkkonnect)) only support TCP. Tim Cooper, has a [branch of Gumble that supports UDP](https://github.com/layeh/gumble/tree/udp), but neither of the clients officially support switching between UDP and TCP or forcing either. If Gumble code once start supporting UDP, this will probably be reflected in talkiepi and talKKonnect, at such time. But as far as I understand neither Tim, Daniel or Suvir probably wont be going there unless there is some strong motivation. Perhaps there will be more people interested, who may want to contribute to UDP mode in Gumble and then choosing between TCP and UDP would be possible, maybe through an excellent XML configuration method that Suvir had created for setting talKKonnect. Right now there is nothing to force. Barnard, talkiepi and talKKonnect are all TCP only.

zrdimetc, 2020-03-07:
Since Suvir‘s [talKKonnect](https://talkkonnect.com) project is a fork of [talkiepie](https://github.com/dchote/talkiepi) from [Danel Chote](http://projectable.me), adding an incredible set of new features bringing a talkiepi concept to a new level and both talKKonnect and talkiepi were made possible thanks to excellent groundbreaking work of [Tim Cooper](https://layeh.com) who created [Gumble](https://github.com/layeh/gumble) libraries to make Go Mumble clients and both talkiepi / talKKonnect possible, all of the clients based on Gumble ([Barnard](https://github.com/layeh/barnard), [talkiepi](https://github.com/dchote/talkiepi) and [talKKonnect](https://github.com/talkkonnect)) and possibly other originating from Gumble, only support TCP. Tim Cooper, has a [branch of Gumble that supports UDP](https://github.com/layeh/gumble/tree/udp), but neither of the clients officially support switching between UDP and TCP or forcing either. If Gumble code once start supporting UDP, this will probably be reflected in talkiepi and talKKonnect, at such time. But as far as I understand neither Tim, Daniel or Suvir probably wont be going there unless there is some strong motivation. Perhaps there will be more people interested, who may want to contribute to UDP mode in Gumble and then choosing between TCP and UDP would be possible, maybe through an excellent XML configuration method that Suvir had created for setting talKKonnect. Right now there is nothing to force. Barnard, talkiepi and talKKonnect are all TCP only.

zrdimetc, 2020-03-08:
Since Suvir‘s [talKKonnect](https://talkkonnect.com) project is a fork of [talkiepie](https://github.com/dchote/talkiepi) from [Danel Chote](http://projectable.me), adding an incredible set of new features bringing a talkiepi concept to a new level and both talKKonnect and talkiepi were made possible thanks to excellent groundbreaking work of [Tim Cooper](https://layeh.com) who created [Gumble](https://github.com/layeh/gumble) libraries to make Go Mumble clients and both talkiepi / talKKonnect possible, all of the clients based on Gumble ([Barnard](https://github.com/layeh/barnard), [talkiepi](https://github.com/dchote/talkiepi) and [talKKonnect](https://github.com/talkkonnect)) and possibly other originating from Gumble, only support TCP. Tim Cooper, has a [branch of Gumble that supports UDP](https://github.com/layeh/gumble/tree/udp), but neither of the clients officially support switching between UDP and TCP or forcing either. If the Gumble main code starts supporting UDP, this will probably be reflected in talkiepi and talKKonnect, at such time. But as far as I understand neither Tim, Daniel or Suvir probably wont be going there unless there is some strong motivation. Perhaps there will be more people interested, who may want to contribute to UDP support in Gumble and then choosing between TCP and UDP would be possible, maybe through an excellent XML configuration method that Suvir had created for setting talKKonnect. Right now there is nothing to force. Barnard, talkiepi and talKKonnect are all TCP only.

zrdimetc, 2020-03-08:
[Question](https://github.com/layeh/gumble/issues/46) was asked to the author of Gumble. As per the author "gumble currently only has TCP support, so no configuration should be needed."

zrdimetc, 2020-03-08:
[Question](https://github.com/layeh/gumble/issues/46) was asked to the author of Gumble. As per the author _"gumble currently only has TCP support, so no configuration should be needed"._

closed 2020-04-24 by talkkonnect