talkkonnect/talkkonnect #18: Error when following the README...

nibor-b, 2020-06-11:
```
/home/talkkonnect/gocode# go get github.com/talkkonnect/talkkonnect 
unexpected directory layout:
	import path: _/home/talkkonnect/gocode/src/github.com/hegedustibor/htgo-tts/handlers
	root: /home/talkkonnect/gocode/src
	dir: /home/talkkonnect/gocode/src/github.com/hegedustibor/htgo-tts/handlers
	expand root: /home/talkkonnect/gocode
	expand dir: /home/talkkonnect/gocode/src/github.com/hegedustibor/htgo-tts/handlers
	separator: /

```

talkkonnect, 2020-06-13:
The Error is from the library github.com/hegedustibor/htgo-tts which is a library outside which is out of my control. Let me copy the library into my github repo and try to fix it.

talkkonnect, 2020-06-13:
I have vendored the problematic library, seems the author updated it a week ago and something is broken. So I copied the working copy into my repository and the go get will pull from my repo, Tested it and it works now. Thank you for the feedback!

talkkonnect, 2020-06-27:
Since many users has confirmed that the instructions in this newly updated README.md file is working as expected. I will close this ticket.

closed 2020-06-27 by talkkonnect