talkkonnect/talkkonnect #20: Asamixer set volume failed

MirkoUgoliniDev, 2020-09-05:
Hi Suvir,

I got this error in log

2020/09/04 08:49:30 PlayWavLocal(EventSoundFilenameAndPath) Returned Error:  alert: set volume failed: failed to execute "amixer set Speaker 100%" (exit status 1)

but executing the same command in shell it works.

[image]

Regards

Mirko Ugolini

talkkonnect, 2020-09-05:
Mirko,
Can you try to run talkkonnect as root and make sure pulse audio is not installed. Is this happening only in the latest build? Or is this something new?

regards
suvir

MirkoUgoliniDev, 2020-09-08:
Hi  Suvir,
thanks for the quick response.

Pulse audio is not installed.
I run  as root and  I Just installed  it some day ago, so I do not  know  if the problem was present in  previous releases.

I got a question...

I would like do  handle the project  with Visual studio Code  plus  Go plug in but I got  different problems  building it

do you use VS Code  too?

Regards

Mirko

talkkonnect, 2020-09-09:
Hi Mirko,
I developed talkkonnect in nano on raspberry pi directly. However I later learned to use VS Code and the way I remotely develop using vs code is to use a package for windows called sshfs to mount the folder /home/talkkonnect/gocode/src/github.com as a shared drive and then add that to VS Code Project. With this setup you can develop on real raspberry pi and everything works as expected.

to compile I have a bash script on the rapsberry pi like the code below 

export GOPATH=/root/gocode
export GOBIN=/root/gocode/bin
cd $GOPATH
cd $GOPATH/src/github.com/suvirkumar/talkkonnect
go build -o /root/gocode/bin/talkkonnect cmd/talkkonnect/main.go

This is how I work and it seems convenient for me do let me know if this works for you. For now talkkonnect has only been tested on linux and works on raspberry pi and also orange pi. Also works on minimal debian on pc running linux.

best,
suvir kumar

MirkoUgoliniDev, 2020-09-09:
Hi  Suvir,

I mapped  with sshfs  the  raspberry's source code  and with  VSCode  I'm able to editing  the code , but  for build  it I have to run the  scrit directly  on  rasperry. 

do you the same?

Regards

Mirko

Mirko

talkkonnect, 2020-09-10:
Hi Mirko,
yes the compile script is on the raspberry pi as a bash shell, you can have a ssh window open below VS code and run the script when you want to compile your code. 

That is how I do it currently if I am using VS Code for that programming session. Don't know if there is a better way to do it or not however that is how I compile now.

best,
suvir

closed 2020-09-17 by talkkonnect