talkkonnect/talkkonnect #21: Segmentation Violation

ChZeman, 2020-09-13:
I'm not sure if this has anything do with my soundcard errors (which I'm working on, audioinjectorpi), but this is what I get when starting talkkonnect:

`[  info ] 2020/09/12 21:29:40 talkkonnect.go:59: Talkkonnect Version 1.47.21 Released September 12 2020
[  info ] 2020/09/12 21:29:40 talkkonnect.go:60:                               
[  warn ] 2020/09/12 21:29:40 client.go:277: Unable to Unmute  failed to execute "amixer set Master unmute" (exit status 1)
[  info ] 2020/09/12 21:29:40 client.go:702: Connected to talkkonnect-community-server Address 149.28.123.3:64738 on attempt 0 index [0]
 
[  info ] 2020/09/12 21:29:40 client.go:705: Welcome message: 
[  info ] 2020/09/12 21:29:40 client.go:707:                                   
[  info ] 2020/09/12 21:29:40 client.go:707:  Welcome to this server running Murmur.
[  info ] 2020/09/12 21:29:40 client.go:707:  Enjoy your stay!
[  warn ] 2020/09/12 21:29:40 client.go:807: PlayWavLocal(EventSoundFilenameAndPath) Returned Error:  alert: cmd.Run() for /usr/bin/aplay failed with exit status 1
[  info ] 2020/09/12 21:29:40 client.go:822: Current Channel  Root  has ( 2 ) participants
[  info ] 2020/09/12 21:29:40 client.go:1107: 1. User "RP_Repeater" is online. []
[  info ] 2020/09/12 21:29:40 client.go:1107: 2. User "RP" is online. []
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x42bd08]

goroutine 25 [running]:
github.com/talkkonnect/go-oled-i2c.(*Oled).sendOledCommand(...)
	/home/talkkonnect/gocode/src/github.com/talkkonnect/go-oled-i2c/oled.go:220
github.com/talkkonnect/go-oled-i2c.(*Oled).DisplayOff(0x0, 0x6, 0x1)
	/home/talkkonnect/gocode/src/github.com/talkkonnect/go-oled-i2c/oled.go:247 +0x34
github.com/talkkonnect/talkkonnect.(*Talkkonnect).BackLightTimer.func1(0x228e000)
	/home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/client.go:2432 +0xf0
created by github.com/talkkonnect/talkkonnect.(*Talkkonnect).BackLightTimer
	/home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/client.go:2415 +0xec
root@raspberrypi:/home/talkkonnect/bin# 
`

Thank you,
Chris

talkkonnect, 2020-09-13:
Hi Chris,
do you have aplay installed? Try to run it from command line using the command /usr/bin/aplay somefile.wav
If you do not have it do an apt install alsa-utils

Also use alsamixer to find the name of your sound device and change it in XML config to PCM or Speaker or what ever shows up on the alsamixer as your primary device for audio out for me it is <outputdevice>Speaker</outputdevice>

Let me know now you get on or how you finally manage to solve this problem.

best,
suvir

ChZeman, 2020-09-13:
Hi Suvir!

Thank you for the information! I changed to a USB sound card and that part works now. I made sure OLED is disabled in the XML config (it was), but I'm still having the segfault.

Thank you,
Chris

ChZeman, 2020-09-13:
I got past the problem by commenting out the lines:
"b.LEDOff(b.BackLightLED)" and "Oled.DisplayOff()" in client.go.

"oled enabled" is set to false in my config.

talkkonnect, 2020-09-13:
Chris,
glad to know talkkonnect is started.
Should you have any questions or want to voice chat with us you can join or community server mumble.talkkonnect.com:64738 use any username and use password talkkonnect.

Thanks,
suvir

ChZeman, 2020-09-13:
Thank you, suvir!

My only challenge left is to get it working with my audioinjectorpi (http://www.audioinjector.net/rpi-hat) instead of a USB sound card.

Thank you,
Chris

closed 2020-09-13 by ChZeman