talkkonnect/talkkonnect #22: Password for talkkonnect

bhamrick2, 2020-10-03:
Hello, I have worked through the installation instructions twice now, and both times when I run the command "./talkkonnect" I get an alert saying there was a problem opening talkkonnect.log. It then says "file error: open /var/log/talkkoonnect.log: permission denied"

I then tried running "sudo ./talkkonnect" and it asked me for the password for the talkkonnect user. This got me very confused because I was under the assumption that there should be no password as it was disabled when creating the user. I deleted the user and uninstalled go, started over, and ended up stuck at the same point. I am not entirely sure what I am missing here. Let me know if you need more information and I can provide it!

I am running Raspbian on a Raspberry Pi 3B+

talkkonnect, 2020-10-12:
Dear Bhamrik2,
sorry for slow reply. You need to make sure you create the log file before you run talkkonnect. Also if it is already there then you have some problem in your XML syntax. Some tag is not closed properly in the accounts area. You can verify if your xml file is sane by using some online XML checker. Do let me know how you get along. 

Happy talkkonnecting!

suvir

bhamrick2, 2020-10-14:
Hi Suvir,

That did the trick! Thank you!

talkkonnect, 2020-10-15:
Ok then I will close this ticket.

closed 2020-10-15 by talkkonnect