talkkonnect/talkkonnect #23: Error when go get github.com/talkkonnect/talkkonnect - missing AL/al.h ?

andrewhearn, 2020-10-21:
Hello,

I'm looking forward to using this project for a stage intercom system - I have an Orange Pi Zero and Armbian's Buster image running on it...

However, going through the readme instructions, I get as far as the go get, and am getting the following error:

```
$ go get github.com/talkkonnect/talkkonnect 
go build github.com/talkkonnect/gopus: invalid flag in pkg-config --libs: /home/talkkonnect/gocode/src/github.com/talkkonnect/gopus/opus-1.3.1/.libs/libopus.la
# github.com/talkkonnect/go-openal/openal
In file included from src/github.com/talkkonnect/go-openal/openal/alcCore.go:24:
./local.h:5:9: fatal error: AL/al.h: No such file or directory
 #include<AL/al.h>
         ^~~~~~~~~
compilation terminated.
```

andrewhearn, 2020-10-21:
...this looks like I was missing libopenal-dev !

zrdimetc, 2020-10-22:
Hi Andrew. When you are at OpenAl, may we offer another useful tip? If your Orange Pi Zero build starts producing "Mickey Mouse" audio at any time, make changes in OpenAl config file /etc/openal/alsoft.conf to use 48000Hz audio instead of default setting 44100Hz, to avoid resampling. frequency = 48000. Some more info about [building talkkonnect with Orange Pi Zero](https://talkkonnect.com/2020/04/21/tips-for-talkkonnect-building-with-orange-pi-zero/). Happy "talkkonnecting".

closed 2020-10-26 by talkkonnect