talkkonnect/talkkonnect #24: Error build  by talkkonnect

fft01, 2020-10-21:
Hello,

when i build the talkkonnect program i get the following error:

BackLightTime.Reset undefined (type * time.Ticker has no field or method Reset)

What am I doing wrong?

Kind regards
Dirk

zrdimetc, 2020-10-22:
Hi Dirk. You are correct. We have seen this problem before in the talkkonnect community. Kindly please refer to an excellent [README](https://github.com/talkkonnect/talkkonnect/blob/master/README.md) document from Suvir. He put a lot of effort into documenting [talkkonnect](https://talkkonnect.com/). Solution to the problem you are reporting is ...You need to update your [go version](https://golang.org/dl/) to at least v1.15. Currently [v1.15.3](https://golang.org/dl/). Regards.

talkkonnect, 2020-10-26:
Yes please follow the instructions in the blog do not use apt-get to install golang.

closed 2020-10-26 by talkkonnect