talkkonnect/talkkonnect #25: voiceactivitypin comes on correctly, but stays on

andrewhearn, 2020-10-27:
I have talkkonnect working, and am trying to have an LED light when someone is talking, using the voiceactivitypin GPIO pin.

I'm testing with two clients, talkkonnect on a Pi3 and Mumble on OSX.
When talkkonnect starts up, the LED is off, as expected.
When my other user transmits, I see it's logged `stream.go:209: Speaking-> andrew` and the LED comes on. However, when the user stops transmitting there is no log saying it's stopped and the LED doesn't turn off.
The second client is Mumble on OSX, and that client is showing that it's stopped transmitting.
The LEDs also stay on after I shutdown TalkKonnect.
The participantsledpin LED works fine as expected.

Is this a known bug or is there something I'm missing?
Thanks again!

talkkonnect, 2020-10-29:
This is a known bug. I have to add another timer to switch off the led. However there is a work around for the moment is to use the same gpio pin as the heartbeat as the rxpin. The heartbeat will turn off the led. I know its not ideal but currently thats what we use until I further clean up the code, when time permits.

thank you for your feedback,
suvir

andrewhearn, 2020-10-29:
Thanks, I'll use that for now - works ok now!

closed 2020-10-29 by andrewhearn