talkkonnect/talkkonnect #27: Suggestion: Alert button/LED

andrewhearn, 2020-12-04:
Here's an idea that I don't think is possible at the moment - many theatre type comms units had a push button which will light up an LED on all the other units. This is useful to get attention of your colleagues in case they don't have their headset on.

It would be good to add this an an option to Talkkonnect - I'd imagine an option for selecting which GPIO pin for the push button switch, and which GPIO for an LED. Then whenever anyone presses their button the LED on all the units will light up for X seconds.

Thank you for the consideration.

zrdimetc, 2020-12-06:
@andrewhearn. There are several alerting functions programmed in TalKKonnect. "Away Toggle Switch" currently works with a flip switch to send a message like "Available" or "I am away" to the channel. On a new message, LCD or OLED screen will light up. Screens illumination / backlight led can be set to time out. New message will turn the screen light back on. Instead of "Away" message you can call this _"Everybody stand by for an important announcement"_. Then perhaps, LCD screen or led's could blink too when receiving this type of a message with some changes in `client.go` and `stream.go` and xml config. There is also a panic alert function in TalKKonnect. On a button press, an alert tone can be played. Alert message will be sent to the channel and mic can optionally be turned "hot". GPS position can be sent too. It's basically a call for help! This can be repurposed too for your type of alert. Please give some time to Suvir, he will probably want to introduce a new function to support what you asked for? Do you perhaps use TalKKonnect's in theater or video production? Can you post or send some pictures of your working units? There are many functions in TalKKonnect purposefully built for communicating for safety and security. Another role TalKKonnect can definitely fill is in helping the video production crews or event organizers communicate effectively. The missing components are battery packs... you need to shut down TalKKonnect in a controllable way and manage the battery. Other types of connectivity may need to be supported for such use cases... adding a 3G/4G modem or working over a phone tethering if the site doesn't have a working wireless. TalKKonnect is a fantastic framework that gives people freedom to apply their own communication logic to their projects any way they like. It's like a FreeSWITCH for PTT with GPIO. TalKKonnect has all the functions and signals needed for turning it into Radio over IP gateway. HAM radio operators love that functionality. Best regards,
Zoran.

talkkonnect, 2020-12-07:
@andrewhearn thank you for your suggestion, have talked with zrdimetc over this request and we do think this is a good idea to have an attention calling LED on the client. When I have some time I will add this feature and I plan to add this feature by adding a specially crafted text message sent via mumble to the all clients. If this message is received then there will be a logic in the XML settings as to which group each user belongs to. This means you should be able to turn on the attention LED of individuals, groups or all users in the channel. I think it would also be a good thing if in the XML file for each user they can set how they would like to be notified, meaning the behavior of the LED, blinking or on for how many seconds depends on each user preference how he would like to be alerted.

andrewhearn, 2020-12-07:
This sounds great thanks!

I am using talkkonnect for live streaming services from our Church. We have people in different areas of the building who need to talk to each other, and we're finding talkkonnect to work really well. Currently I have 2 OrangePi Zero boards, a mumble client on a laptop and a Raspberry Pi running the server, but will be adding more Talkkonnect clients when my Aliexpress order arrives! I'm using either 'gaming style' headsets with USB connection or USB sound card devices. To save costs and to make the units simple I'm not using the oled screen, but have a couple of push switches and LEDs. I'll post photos when the new units arrive!

closed 2020-12-16 by talkkonnect