talkkonnect/talkkonnect #28: COR/COS/Carrier Detection/Busy LED not working

4d69636b, 2020-12-07:
I'm using TK as an RF gateway to UHF CB (477MHz FM) in Australia. The busy LED (COR/COS/Carrier Detection) is used to drive the RF transmitter on. 

Version: 1.52.02

Issue: When a mumble user/client starts to talk the RF gateway key's up but then drops off. Only re-keys up if the mumble user un-keys and re-keys their client PTT and the issue repeats.

I have moved back to version 1.50.02 which works but RF gateway stays keyed with a short tail (<1second)

zrdimetc, 2020-12-07:
Hi, you didn't provide info about the exact interface circuit to key up your transceiver PTT and your board? talKKonnect will work with any type of a radio, as long as your PTT interfacing method is adequate. RX Led in talKKonnect version 1.52.02 work as expected. **You need to use a different GPIO for heartbeat or disable the heartbeat functionality!** In earlier versions of talKKonnect it was possible to use the same GPIO for both heartbeat and RX activity. Not any more. Suvir created a clever new function for controlling RX Led. It works as a software "capacitor". For radio-gateway operation you can increase `TimerTalked` value in `stream.go` file from 200ms to 1000ms or higher, depending on the quality of your network. With lossy networks and very short ticker values, you may experience some Rx Led blinking, which you would like to avoid in a gateway. Set the `TimerTalked` to a higher value if you experience RX Led flickering with packet loss? `TimerTalked` will be configurable through xml config later on when Suvir will have more time. For interfacing to the radio, you need to drive a transistor, mosfet, an opto-coupler, a small relay from the RX Led GPIO. You can only draw a very low current from GPIO, like the Milliampers, 10-15 mA or so to power up LED or drive a base of a transistor. 3.3V rail usually supplies a maximum of 50 mA and individual pins current maximum can be even lower, like 16mA. If a radio PTT interface is drawing a higher current, then it is possible to experience a significant voltage drop in the GPIO output. What is your board? Test and verify GPIO operation with some other program? Measure a voltage on RX Led GPIO when you key a radio? Is it anywhere near 3V? Current from GPIO? If you can ensure that your PTT input in the radio is galvanically isolated from the GPIO through an interface with transistor and optocoupler, this would be ideal. Go back one step and test talKKonnect version 1.52.02 without a connection to the radio first? Is the RX led functioning properly? If not, disable the heartbeat or ensure heartbeat and RX led are different GPIO. If RX Led is still not working, review your PTT interface carefully. PTT interface can be a very basic circuit. Recommend transistor to drive optocoupler. Let's say 2N2222 or 2N3904 and PC817. OC then controls the radio PTT. If your PTT interfacing method is correct, the most likely reason for your issue is a conflict between heartbeat function and new RX Led function. Let us know if the issue is fixed? Can you send a few pictures of your radio gateway to Suvir? Radio-gateway is one of the most important features of talKKonnect and one of the reasons behind Suvir's design. Regards, Zoran

4d69636b, 2020-12-10:
Thank you for your detailed reply, Zoran.

I have **disabled the heartbeat** and the RF radio now transmits fine. As you pointed out the RX activity follows the current speaking user much more closely. I'll make use of the TimerTalked delay if need be.

Thanks, Mick

closed 2020-12-10 by 4d69636b