talkkonnect/talkkonnect #30: Feature request

umekzg, 2021-01-06:
Is it possible to implement Whisper/Shout to channel/user function that exists in gumble (VoiceTarget function)?

talkkonnect, 2021-01-07:
Since talkkonnect runs on raspberry pi and was designed to have a desktop form factor with LCD or OLED screen, there is no way to select the user that you would like to whisper to. So the short answer is that whisper is not implemented as there is no good way to do it with the current form factor and user interface, unlike when running mumble on a PC/MAC/Android where you can click on the username and whisper to him/her.

The whisper/shout function was also not fully implemented in the gumble software library.

Happy Talkkonnecting!

umekzg, 2021-01-07:
It would be a big feature to have the option to whisper to a set of predefined channels with GPIO triggering.
Like a big expensive talk-back intercom system.
Nevertheless thank you for your answer.

talkkonnect, 2021-01-08:
Zoran and I tested this whisper/shout feature. It works if you register the user on the server (on talkkonnect receive), however for transmit there is a gumble library file called voicetarget.go let me play around with that when I have some time out of my crazy work schedule. I will add it to my todo list.

thank you for the feature request and for your use case where you want a talk-back intercom system, I did a little research and found it is used in recording studios which is an interesting use case. 

Give me some time and I will test it and if it works I will release it in the future.

Happy Talkkonnecting!

closed 2021-01-10 by talkkonnect

chalamministries, 2021-07-12:
I second this feature. Here's basically what I'm trying to build. The issue is they run a light desktop and use keyboard shortcuts. But look at the base station where they have 4 LCD screens and 8 ptt buttons. 

[https://inside.arcada.fi/kultur-och-media/made-in-arcada-an-open-access-diy-intercom-system/](https://inside.arcada.fi/kultur-och-media/made-in-arcada-an-open-access-diy-intercom-system/)

talkkonnect, 2021-07-17:
Yeah I see that but for that to work I would need to get the shout and whisper function working also completely refactor the display Aung GPIO logic. With the current situation I don’t have a lot of time and motivation to make all these changes. Hope this COVID times pass soon.