talkkonnect/talkkonnect #31: Joining passcoded channel/sub channel from startup

balkan16, 2021-02-04:
Hi!! Thanks for the project, have been looking for this exact project for the past 2 months. It really helped me a lot. I've been wondering could the **raspi(talkkonnect)** join a **specific passcoded channel/sub-channel from the startup**? I've been looking through the XML code and the tutorial. Seems there's **no specific place to tell the password of the channel**.

talkkonnect, 2021-02-04:
In the account tag there is an xml element called channel

balkan16, 2021-02-05:
I've already done that, the problem is when the channel/sub-channel has a password on it. How do i insert the password  so i could automatically join that "secured channel"?

talkkonnect, 2021-02-05:
Have never tested with a secure subchannel before. Question does this work if you connect to the root channel first then change channel? Or does it also not work that way? Just want to know cause never tested to connect to a secure channel. Will have to look at humble library documentation.

balkan16, 2021-02-05:
From root, then move to another channel(without password) works perfectly. But if there's a password, the only choice i have is to move talkkonnect to that particular channel as an admin. 
There's also a problem which **admin couldn't register talkkonnect**, if the admin could register talkkonnect, it wouldn't have to go back to Root again after every startup.

talkkonnect, 2021-02-05:
According to the docs I have read and remember, this may be inaccurate, but it seems all client "Have to" connect to the root channel first, they they are moved after connection. This is what happens in the gumble library. That is why I think when you connect to a server with normal password and move to a normal channel that has no special password for that channel it works. What I am thinking is that maybe something is broken with the password authentication on talkkonnect or maybe we are not following the mumble protocol rules for password protected channels. At this point I really don't know the cause but if you can figure it out do let me know. When I have time later I will look into this issue, as i have never tried this use-case before. Maybe reading the mumble protocol documentation will give some idea how to solve this problem and how authentication for individual channels works in mumble.

balkan16, 2021-02-05:
Sure, I appreciate your effort! Thanks for the response.

talkkonnect, 2021-02-05:
I think you may have to use a certificate for that. Can you try to use a certificate and let me know if it works for our use case? Without the certificate I don't think you can register a user on a channel. So its something to do with requiring a certificate. Let me know how you get a log. As for how to generate a certificate there are some instructions in the talkiepi page.

balkan16, 2021-02-06:
Yes, the certificate works perfect, now I could register talkkonnect and automatically assign it to a specific secured channel after every startup! Thanks for the solution.

closed 2021-02-06 by balkan16