talkkonnect/talkkonnect #32: Error Message(Can't talk or hear)

balkan16, 2021-02-06:
So i got this message,

`[  info ] 2021/02/06 20:04:30 Message (105) from  strong-your-client-doesnt-support-the-celt-codec-you-wont-be-able-to-talk-to-or-hear-most-clients.-please`

I couldn't transmit or receive anything. Any clue why is this happening?

talkkonnect, 2021-02-08:
This is a murmur server side setting, talkkonnect only supports OPUS codec. However one of the persons connected to the server is using an older codec (possibly CELT) and it is not forced on the server to use OPUS to the server goes back to support older codec for all the users for legacy support. The setting on the server should be opusthreshold=0 (in the murmur.ini file) for the server not to switch back to older CELT support. All mumble clients support OPUS there is no need to go back to CELT anymore.

balkan16, 2021-02-09:
I've checked opusthreshold on my server, it's already set to 100. I tried to reinstall talkkonnect and it finally works fine.

talkkonnect, 2021-02-09:
It should be zero not 100

balkan16, 2021-02-10:
Oh sorry, my bad. So the error actually keeps coming back, reinstalling the talkkonnect will only get rid of the error for a temporary time. I decided to change the opusthreshold to 0 on my server and the error still the same.

talkkonnect, 2021-02-11:
The problem for codec support should be solved if you have changes the opusthreshold to 0 and restarting the mumble server.

closed 2021-02-11 by talkkonnect