talkkonnect/talkkonnect #34: Talkkonnect Autostart(Already installed Pre-made Image)

gaban20, 2021-03-12:
Hi, I'm having auto startup problem with the pre-made image, I have followed your instructions about the startup script. After i reboot/turn on, it keep asking user and its password.

talkkonnect, 2021-03-13:
Which user and password? if its the linux username and password the user is root and password is talkkonnect. Please elaborate.

gaban20, 2021-03-13:
I'm sorry if I'm not clear enough. I'm trying to use your pre-made image for gateway for ham-radio. I was expecting that your pre-made image could be run by itself since startup. After i checked from my monitor, it turns out that the raspberry asking for username and password even after i have followed your instructions about making the startup script(talkkonnect-run). What should I do to override the login process to Root?(I'm assuming the talkkonnect only works with Root user)

talkkonnect, 2021-03-14:
After installing talkkonnect runs as a service. This means it runs in the background you dont need to interact with it. However if you want to interact with it you can log in and type screen -r as it is running in a screen.

gaban20, 2021-03-16:
I reinstall the image and it works fine just like you said after changing to my server. I noticed a problem which happen when i change username/certificate in the talkkonnect.xml. The autostart not working anymore even after i change it back like before. Do you happen to know what's wrong with it?

talkkonnect, 2021-03-16:
Follow the steps to generate certificate here https://github.com/dchote/talkiepi/blob/master/doc/README.md and add it to the XML file.

gaban20, 2021-03-16:
Already done that. The problem just like i said earlier.

talkkonnect, 2021-03-16:
On Andreas Spiess server I am using the following settings for certificate. Usually for my servers I use username and password without certificates.

For your reference I have attached my config that works for me below

            <account name="fr2.imumble.nl" default="true">
                        <serverandport>fr2.imumble.nl:57264</serverandport>
                        <username>yourname</username>
                        <password></password>
                        <insecure>true</insecure>
                        <certificate>/home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/mumble.pem</certificate>
                        <channel></channel>
                        <ident>Name Surname</ident>

talkkonnect, 2021-03-18:
Since there is no more activity or questions or feedback here allow me to please close this ticket.

closed 2021-03-18 by talkkonnect