talkkonnect/talkkonnect #35: Garbled received audio with two other mumble clients

andrewhearn, 2021-04-06:
Hello! This is a little odd, and I'm wondering where to look to fix this...

I have an Orange Pi Zero running TK, and it works fine when talking with one other Mumble 1.4 client on OSX or Windows. 
However, when a second Mumble 1.4 client on OSX/Windows joins and starts talking the audio as heard on TalkKonnect is all garbled and has some very interesting long delay and echos - this is whilst the audio as heard on the two Mumble clients is fine.

This is all whilst audio from TalkKonnect is heard fine on the windows/osx clients.

The Mumble server us running on a separate Pi4.

All clients are using wired ethernet.

On the same Orange PI, I have also installed Talkiepi and that works fine! 

I've used the same /gumble/audio.go settings from TalkiePi in TalkKonnect and it makes no difference.

This doesn't seem to be CPU bound on the Orange Pi - it has 4 cores and non are maxed out at all when running TalkKonnect.

Has anyone any pointers as where to look to fix this?

Many thanks!

talkkonnect, 2021-04-07:
Quick question just to make sure you are using the current version of talkkonnect version 1.62.02?

talkkonnect, 2021-04-15:
Since there is no activity here in this bug report and also I cannot reproduce this bug I will close this ticket.

closed 2021-04-15 by talkkonnect

talkkonnect, 2021-05-15:
Finally found out why this is happening, this means that 2 or more devices are transmitting over each other causing a break up in audio or causing slow audio. Talkkonnect modified the gumble library to prevent go routine leaks and this was a byproduct of fixing that go routine leak. Talkkonnect will not tolerate people in full duplex mode speaking over each other,

cschriel, 2021-07-13:
We've experienced an issue similar to this one. As soon as a client connects, rPi 4 clients would experience garbeled and delayed audio. Our fix was turning off all sound effects on the rPi's.

talkkonnect, 2021-07-14:
This problem is caused by the client connecting an transmitting and there
are two or more clients transmitting over each other. Make sure only one
client is transmitting at one time and see if the problem persists.

Suvir

Keksstar, 2024-01-04:
Wondering if this has been fixed? I´m hearing something similar. I have one client that "speaks"/is open all the time. As soon as I press PTT on an client the received Audio get´s garbled and sadly doesn´t become clear again after releasing the PTT. If it once has the state, it´s garbled forever until reboot.

talkkonnect, 2024-01-04:
Please try with the latest update. Do the following to update (This works on Raspberry Pi not sure about other boards)
1. log in to the raspberry pi board using root account
2. wget https://raw.githubusercontent.com/talkkonnect/talkkonnect/main/scripts/update-talkkonnect.sh
3. chmod +x ./update-talkkonnect.sh
4. ./update-talkkonnect.sh

reboot and you should have the latest version running for testing.

No One else who is using a board better that raspberry pi zero v1 (old version) has reported any audio issues. Anything from Raspberry Pi 0 2w, Raspberry pi 2, Raspberry pi 3 series, Raspberry Pi 4 and Orange Pi Zero all work in production.

Please note that the images are not updated to the latest version, so even if you use the image you will have to update as of this time.

Keksstar, 2024-01-04:
Thanks for the reply! I did the update and rebootet the Board. (NanoPi Neo Core - great Board, lots of power) Sadly that doesn´t help.

I did an recording straight from my headphones... Around 5 seconds I press the PTT button: You can notice an shift in pitch and audio artifacts starting to occur.
[Recordings.zip](https://github.com/talkkonnect/talkkonnect/files/13835526/Recordings.zip)

talkkonnect, 2024-01-05:
We have tested on Raspberry Pi and Orange Pi, used mostly for simplex communications. I guess your use case may be special so I cant reproduce the same behavior here. People are running talkkonnect for mission critical communications and we have not received any reports of any audio issues. Maybe it has to do with your use case is full duplex, I am not really sure. 

I am also now using version 1.5 and 1.6 of the murmur server on the internet with 100s of users we don't have this problem.

Anyway if you can try on rapberry pi and let me know otherwise I am not really sure how I can help. I am thinking its something with full duplex but not sure cause most use cases do not have full duplex.

There is a another open source called seren and its written in C and it supports full duplex for up to 10 users and works very well with awesome audio. You may want to check that out if it works for your use case.

Keksstar, 2024-01-07:
> There is a another open source called seren and its written in C and it supports full duplex for up to 10 users and works very well with awesome audio. You may want to check that out if it works for your use case.

Sadly I´m unable to find. Could you please give me an link?

talkkonnect, 2024-01-07:
http://holdenc.altervista.org/seren/index.html

Suvir Kumar
Global Coders Co., Ltd.
262/15 Soi Onnut 46,
Onnut SubDistrict, Suan Luang District
Bangkok 10250
THAILAND
Tel. 668 1 840 7160

Keksstar, 2024-01-08:
Problem solved: When using the onboard audio it´s not working, as described above. When using an external USB Card everything is fine - even with fullduplex. 👍

talkkonnect, 2024-01-08:
ok great, thats maybe why i could not reproduce the problem here. Rest assured talkkonnect works well in production.
also we cannot develop and test for non mainstream boards that we dont have so supporting the less popular boards becomes difficult.

Happy talkkonnecting!