talkkonnect/talkkonnect #36: question

qsoMaster, 2021-05-16:
Hello . I tested your project on raspberry pi 3 and orange pi zero, it works fine, but there is one thing, the volume level can be adjusted only in Linux itself (alsamixer), and I did not find such an opportunity in talkkonnect, as for example in the mumble client. It is believed that the sound through the sound card (cm-108) is very quiet and with a higher gain of the sound amplifier for sufficient volume, quantization noises are strongly audible.
Question: Will it be possible to adjust the levels of reception and transmission in the talkkonnect itself.
Of course, thank you so much for your project before I tried asterisk + app_rpt (allstarlink), frn server + client for linux, svxlink (openrepeater). All this is not something there is no client for android and it is possible to control remotely, but it is difficult to implement and insufficiently documented or abandoned.

zrdimetc, 2021-05-18:
Your DC power supply must have impeccable filtering. Try different power supplies and note the diference? Switching converters in them can be noisy. Filtering condensers could be dry. CPU in SBC can be noisy. As you said, quantization in encoding/decoding can produce a noise. You can try other better quality sound cards? Use ferrite beads and shielded audio cables for audio connections? Keep audio cables short? Try some RC filtering? In talKKonnect, you can use F5/F6 for volume up or down and F3 for mute/unmute. Use Del key for a shortcuts menu. For adjusting the audio input level you must use alsamixer, amixer, pactl / pacmd, etc. Pulseaudio can deliver somewhat less noisy audio from Alsa. If your audio input is too low, despite a high adjustment level in audio mixer, try replacing mic bias resistor in CM108 sound card for a lower value? talKKonnect is a great communications framework allowing, among other things, amateur radio operators or other experimenters and enthusiasts to easily create Radio over IP gateways... make hybrid comms networks joining radios with IP networks. But you need to create  radio interface for your type of radio. It's not difficult at all provided you posses basic electronics knowledge. talKKonnect / Mumble channel activity can be used to produce a signal for controlling radio PTT. Radio channel activity can be used to trigger talKKonnect PTT.

talkkonnect, 2021-05-20:
In addition to what zrdimetc said in his comment above, you can adjust the volume using F5,F6 in the talkkonnect console, or remote control via http api using http://{{talkkonnectIPAddress}}:8080/?command=VolumeUp and http://{{talkkonnectIPAddress}}:8080/?command=VolumeDown.

You can also use MQTT topic VolumeUp and VolumeDown and control talkkonnect remotely via MQTT commands from anywhere in the world.

talkkonnect, 2021-05-27:
Since no further questions or responses will close this ticket

closed 2021-05-27 by talkkonnect

qsoMaster, 2021-05-27:
Hello, I'm debugging, I'm using an analog power supply from a GR-300 repeater in conjunction with a motorola radio station, the raspberry is powered through a 5v voltage regulator from a gr-300 PSU. The sound quality from the receiver to the mumble server is good there is no digital noise, when the signal passes from the server to the talkkonnect side of the transmitter (it works on the antenna equivalent of SWR = 1.1) there is digital noise, it is not very strong, but it is there, trying to understand where it comes from.

qsoMaster, 2021-05-27:
F5/F6 for volume up or down - it use alsamikser wol up or down . not in mumble client . I am filtring +5v power on cm 108 chip . noise is down . but still there .
If the volume from the client's mumble were higher, I would crush the noise with a resistor divider to a level that is barely audible, but in this version this is impossible, since the deviation level of the transmitter will be insufficient, I will try another sound card when the package from aliexpress arrives.

talkkonnect, 2021-05-28:
Talkkonnect uses the alsamixer volume there is no separate volume of the application itself. The noise can be cut down by using a good quality sound card and shielding the raspberry pi itself along with using shielded cables on the input and output.

qsoMaster, 2021-05-30:
Today I solved the problem of digital noise at the output of the sound card, as I thought it turned out to be in the sound cards, I found a USB card on a different chip (defined as JMTek, LLC) with it, everything works great. That's what's inside her [https://cxem.net/sound/raznoe/usb_mic.php](url)

talkkonnect, 2021-05-30:
Can you please take a picture and give more details so that others including me can get the same sound card. This will help others with the same problem.

qsoMaster, 2021-05-30:
# This card is defined as:ID 8086: 0808 Intel Corp.
lsusb:

Bus 002 Device 007: ID 8086: 0808 Intel Corp.

During operation, there is a strong digital noise at the output.

[image]

# This card is defined as:ID 0d8c:0012 C-Media Electronics, Inc.
lsusb:

Bus 001 Device 004: ID 0d8c:0012 C-Media Electronics, Inc.

Low digital noise at the output during operation. 

[image]

# This card is defined as:ID 0c76:161f JMTek, LLC. Work Fine ! no noize .
lsusb:

Bus 002 Device 006: ID 0c76:161f JMTek, LLC.

Work Fine. 

[image]

talkkonnect, 2021-06-04:
Thank you for your information this will be useful for those who face noise problems when interfacing with Radio Transceivers.