talkkonnect/talkkonnect #39: can not communicate with wpa_supplicant

spereg53, 2021-07-16:
i downloaded mems microphone image for raspberry  and can not connect to wifi,after sudo raspi-config ,wireless lan and give the message that can no comunicate with wpa_supplicant,wha can i do???

spereg53, 2021-07-17:
[image]
[image]

talkkonnect, 2021-07-17:
Are you running as root you should be running as root

talkkonnect, 2021-07-18:
also with googlefu other users using raspberry pi solved the same problem by using the following command to debug

sudo wpa_supplicant -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlan0 -d

zrdimetc, 2021-07-18:
Edit your `wpa_supplicant.conf` file

`nano /etc/wpa_supplicant/wpa_supplicant.conf`

Add your own wireless network information

```
network={
ssid="your_ssid_name"
psk="your_wifi_password"
}
```

Restart wpa_supplicant service

`service wpa_supplicant restart`

spereg53, 2021-07-18:
thanks ,i did modify wpa_supplicant manual mode
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

closed 2021-07-19 by talkkonnect