talkkonnect/talkkonnect #40: Go Mod support

s00500, 2021-07-18:
Hi, this is an awesome project!!!

Could you add go module support to the main project ? it should not be too complicated, if you want I can submit a PR

Greetings,
Lukas

talkkonnect, 2021-07-19:
Hi Lukas,

Thanks for your feedback and I appreciate that you like the project. I am self taught and still have a lot to learn about programming.

I followed the instructions on https://stackoverflow.com/questions/55664630/how-do-i-migrate-from-dep-to-go-modules
to migrate to go modules, but so many of the tests failed because of the libraries I vendored.

I managed to create a file go.mod with contents like this in the /home/talkkonnect/gocode/src/github.com/talkkonnect
I am not sure if it is correct pls advise.

module github.com/talkkonnect

go 1.15

require (
        github.com/comail/colog v0.0.0-20160416085026-fba8e7b1f46c
        github.com/davecgh/go-spew v1.1.1
        github.com/eclipse/paho.mqtt.golang v1.3.5
        github.com/glendc/go-external-ip v0.0.0-20200601212049-c872357d968e
        github.com/go-ole/go-ole v1.2.4
        github.com/golang/glog v0.0.0-20210429001901-424d2337a529 // indirect
        github.com/golang/protobuf v1.5.2
        github.com/itchyny/volume-go v0.2.1
        github.com/jacobsa/go-serial v0.0.0-20180131005756-15cf729a72d4
        github.com/kennygrant/sanitize v1.2.4
        github.com/kidoman/embd v0.0.0-20170508013040-d3d8c0c5c68d
        github.com/mattn/go-runewidth v0.0.13
        github.com/stianeikeland/go-rpio v4.2.0+incompatible
        github.com/talkkonnect/go-nmea v0.0.0-20210719131409-1f6a4b9c6894
        github.com/xackery/gomail v0.0.0-20150929200936-cc84600edc17
        golang.org/x/sys v0.0.0-20210630005230-0f9fa26af87c
        periph.io/x/periph v3.6.8+incompatible
)

Been working mostly alone on this project and I would be very happy to accept code from other developers so I can learn more from others too.

Best,

Suvir

s00500, 2021-07-19:
Hi Suvir, awesome, I also ended on the problem than many of the vendored libs have naming problems, I think either these will need to be fixed or we can work with the replace directive... I will try to make this work and let you know once I got it :-)

Also I think it could be interesting to make an alpine linux based RPI image, maybe I can have a go at that as well...

Greetings,
Lukas

talkkonnect, 2021-07-19:
Dear Lukas,
The go mod would be awesome, some clean up would be required that would also allow us to run tests etc. As I said earlier I am amateur programmer so would wait for your PR.

We are mostly standing by on the community server with our talkkonnects you can drop by for a voice chat at anytime to exchange ideas.

What is your use case for talkkonnect?

Thanks again for your interest in talkkonnect,

suvir

s00500, 2021-07-19:
I have created a draft PR here, 

This PR: https://github.com/talkkonnect/volume-go/pull/2
will need to be merged first, then I can finish it

Ok cool, yeah I've worked quite a bit with go over the last couple of years, that is why this project had my interest fast

I am actually searching for a solution to doing video production intercom, a raspberry pi that has some simple buttons and an oled seems to be just the right fit.

I have finally realized how you are using talkkonnect in general and the concept behind it. As it happens I am also a licensed HAM radio operator, but I am usually not very active on any of the bands, I am mostly interested in electronics and developing stuff, but for having a chat about all this with you I will definitely come back to your offer :-) Lets try to make this run first then =D

As for dep: I don't really see you using it here,  so there is no need to migrate I think. Go has taken some time to develop a good system for managing dependencies, with Go modules they now have one that is solid and easy to use.

To add the two files I basically just ran:
go mod init github.com/talkkonnect/talkkonnect # initialize the project as a module
go mod tidy # automatically create the list of dependencies and download them

doing so also decouples the project from the old concept of the GOPATH, now the folder with all the files can live anywhere on a system, making it more portable.

Greetings,
Lukas

talkkonnect, 2021-07-20:
Dear Lukas,
Merged as per your request.  You can proceed to the next steps. There are many requests to make a system with multiple screens and PTT buttons for private voice messages. The screens we can achieve with a i2c chip and for the gpio for tx and leds maybe we can use some hardware multiplexer.

I have tried to make talkkonnect a platform whereby people can build upon and customize to their own needs. From working with talkkonnect there are 3 scenarios that people are interested in 

1. Use as a Internet Intercom/Radio
2. Use as repeater to bridge between internet and radio to bridge the gap between internet and radio with full gpio control
3. Use as a device for the director of video production to talk to the camera men in the field
The 3rd use case would need to implement whisper and shout features and would replace those systems that cost a lot of money and are non customizable

Best,
suvir

s00500, 2021-07-20:
Can you merge this ? https://github.com/talkkonnect/volume-go/pull/2

talkkonnect, 2021-07-20:
merged pls check

zrdimetc, 2021-07-20:
A natural addition to talKKonnect project, for creating a "video production" fork, would be adding a support for I/O port "expander" chip like [MCP23017](http://ww1.microchip.com/downloads/en/devicedoc/21952b.pdf) . That will allow connecting up to 8 x i2c Oled displays like 0.96” or 1.3“ SSD1306/ SH1106 for a master intercom unit. Another MCP MCP23017 can be used for connecting more buttons and leds. As Suvir pointed out, Mumble "whisper" function equivalent to "private", 1:1 call can be implemented. For this to work clients need to be "registered" first in Mumble server database, make the user name permanent. Once this can be made to work "shout" function equivalent to "group call" in radio terms, call can be implemented too. Integration with some sort of keypad or keyboard for texting from talKKonnect console or through api would also come handy.  Suvir has created a great framework that put's original Gumble and talkiepi on "steroids", when it comes to real-life practical applications. Great work Suvir and keep it up!

talkkonnect, 2021-07-24:
Added go mod support by Lukas, thanks for the pull request and the modifications!
Allow me to close this ticket now

closed 2021-07-24 by talkkonnect