talkkonnect/talkkonnect #43: how to configure mumble for listen all channels

spereg53, 2021-07-21:
how to configure mumble client or server to make that one client listen all channels ,like a hierarchical channel structure????thanks in advance

spereg53, 2021-07-21:
ejample:Use case would be: Talking only on single channel but listening to multiple

talkkonnect, 2021-07-21:
You need to use the shout feature. The client doing the shouting has to shout on a a channel that is the parent channel of the branch. All client children that are on the same branch will hear the shout. Talkkonnect can be a listener but not a shouter at this time. For shouting will need official mumble client. Also the shouter needs to be authenticated on the server first. To be authenticated has a special meaning in mumble. Read the official mumble documentation for shout and whisper functions and how to implement them.

talkkonnect, 2021-07-25:
We are studying the voicetarget part of the gumble protocol, so far we are not able to get it to work yet. These functions work with registered users only. So far there is no way to register a user on talkkonnect so we will look into all this and it will take some time to figure it all out and get it working in practice.

Time permitting we will work on these features.

best
suvir

talkkonnect, 2021-07-27:
Managed to play with the gumble and found out how to whisper next will play with shout. Anyway for the impatient this is how i got whisper to zoran's laptop working. Code excerpt below, we will play with it and come up with some clever way for the hardware UI, maybe a separate button for each whisper target configurable in the xml config file.

// to set target(s)
	TargetUser := b.Client.Users.Find("zoran-laptop")
	vtarget := &gumble.VoiceTarget{
		ID: 1,
	}
	vtarget.ID = 1
	vtarget.AddUser(TargetUser)
	b.Client.VoiceTarget = vtarget
	b.Client.Send(vtarget)

	// to clear target(s)
	//	b.Client.VoiceTarget.Clear()
	//	b.Client.Send(vtarget)
}

talkkonnect, 2021-08-01:
We are working on this and will have the whisper and shout function available in the near future. Allow me to close this ticket.

closed 2021-08-01 by talkkonnect

Keksstar, 2024-05-07:
Wondering if it´s possible to listen to multiple channels at the same time and than talk to selected ones? (just like an real intercom system)