talkkonnect/talkkonnect #44: how can i set the audio input to be the monitor of speaker out

retrospectsecurity, 2021-08-01:
how can i set the audio input to be the monitor of speaker out

i want to setup a raspberry pi headless to play audio from command line into mumble but i cant set audio in to the monitor of audio out. any help?

talkkonnect, 2021-08-02:
You can use Stream Function in talkkonnect on that. See if stream function will work for you.

retrospectsecurity, 2021-08-02:
stream mode just starts playing music from File http://prdonline.prd.go.th:8200 Stopped!

want I am trying to do is use the talkkonnect as TTS annunciator. IE I can send text over http that plays back text to speech of the speaker but it wont play on mumble. how can i use the stream function to play back all sounds on the output not just the stream

retrospectsecurity, 2021-08-02:
is there anyway to play wav file from api? i can accomplish my goal this way if its possible

talkkonnect, 2021-08-03:
So what you are saying is you want to play a file locally to the speaker and command it to start playing remotely over api, correct? Currently talkkonnect cannot do that. However you can install mpd and command it remotely.

retrospectsecurity, 2021-08-03:
I did it. I turn on repeattxloop with api and replace Image.wav. After it plays I delete the wav file and wait for new one to be made.

talkkonnect, 2021-08-03:
Ok glad to know you managed with a work around still don’t understand what your use case is. Can you let me know how you are using this feature? In what situation?

retrospectsecurity, 2021-08-03:
I am using it as a mumble-TTS (text to speech)

I have a separate software that let's me send text over web hook and it saves the speech file as a wav. I then replace the wav file that the api cmd repeattxloop to play back my custom speech.

This can be used for any integration-Ie server down, internet down, whatever you want.
Audible text alerts.

talkkonnect, 2021-08-03:
Ok so basically what you want and have achieved is send some text over http to talkkonnect for talkkonnect to convert to speech by text to speech and then play that sound as an alert over the mumble stream to mumble clients on that channel as some sort of alerting system. Sounds interesting in the future I will add that feature.

retrospectsecurity, 2021-08-03:
Pico2wav is used to generate the wav.
Would be cool as official feature one day. Thanks

talkkonnect, 2021-08-06:
OK All Noted As Considered as Additional Future Feature

closed 2021-08-06 by talkkonnect