talkkonnect/talkkonnect #45: Auto Reconnect

retrospectsecurity, 2021-08-08:
Is there any function to auto reconnect? Say the network goes down and then comes back up - it does not seem to reconnect automatically. any tips?

talkkonnect, 2021-08-08:
If you use our image we have created systemd scripts to restart talkkonnect if it dies for any reason. We use screen to run talkkonnect in the background as a daemon and systemd takes care and makes sure it’s always running.

retrospectsecurity, 2021-08-09:
the issue isn't that the software will restart in the screen its the issue that if network disconnects the software stays running and does not reconnect automatically

talkkonnect, 2021-08-09:
Pls elaborate does this mean the physical Ethernet goes down? Or the internet disconnects? What interface are you using? WiFi or Ethernet?

talkkonnect, 2021-08-14:
No response or feedback for 1 week closing ticket

BeJay8675309, 2022-01-20:
I'd like to comment on this issue. The problem is you lose 4/3g connectivity and the app dies. Screen is still running, but the go.lang just locks up. The only way to bring it back to life is respawn the screen session which if you are waiting for a contact won't happen. the other weird side effect of the crash is the terminal has no cursor until reboot. I made a python script to try and ping the host mubble server and reconnect but it won't killall golang properly and just memory over runs. Seems a legit issue that is easy to replicate if you have a fading internet connection - unlike a "pull the plug" ethernet disconnect.

talkkonnect, 2022-01-20:
How do I replicate this issue I have tried with firewall etc need a sure shot way to replicate pls advise

BeJay8675309, 2022-01-20:
The internet is lost. I'm using a USB 4G dongle that appears as eth1 via linux. So the gateway is still there (the modem) but there is not internet to get to your mumble server) To replicate this I just drive into town where I go through a patch of no 4G coverage and then return. talkkonnect just locks up and won't restart even with your startup script.

talkkonnect, 2022-01-20:
I don’t have a similar setup using WiFi and Ethernet so can’t replicate with the current equipment I have. I will need to replicate the issue so I can see what happens under the hood and try fix it. No good ideas come to mind.

Josef-source, 2022-01-21:
I sometimes have the same problem;  the easiest way to quickly be qrv again is to toggle your device on and off!  With a Huawei HiLink dongle, the connection is available again in a few seconds!

BeJay8675309, 2022-01-21:
Yes that is the issue. However it needs to auto reconnect by itself, otherwise you don't know you are un-contactable in the car.

Josef-source, 2022-01-23:
the problem is with the LTE modem;  These crashes also happen with other programs, especially when there is a cell change or when there is no traffic for a longer time!  I'm currently experimenting with different modem settings;  maybe I can find a solution! 
PS: the Problem dosn't exist, when i use a Huawei Hotspot and TK ist connectet via WiFi with this!  The hotspot also seems to be much more sensitive (LTE) than the dongle!

BeJay8675309, 2022-01-23:
More me it will do it either connected via USB (shows as ethernet interface to linux) or the built in wifi AP inside the dongle. I have tried both a 4G Huawei and a ZTE versions. It also will occur when connected from my hotspot on my samsung phone too. All have the same issue.

zrdimetc, 2022-01-29:
Anyone looking to take talKKonnect to the road, please research “net dialer” function in talKKonnect’s `[clientcommands.go](https://github.com/talkkonnect/talkkonnect/blob/main/clientcommands.go)` code and the methods for creating [system service](https://github.com/talkkonnect/talkkonnect/blob/main/conf/systemd/talkkonnect.service) for talKKonnect. This is important for understanding how talKKonnect is handling starting, restarting and server reconnections. With 3G modem you would normally need to create a [ppp chat script](https://wiki.archlinux.org/title/3G_and_GPRS_modems_with_pppd) or use dialer programs like `[wvdial](url)`, `[sakis3g](https://wiki.archlinux.org/title/wvdial)`, [ModemManager](https://modemmanager.org) or [NetworkManager](https://networkmanager.dev) to handle connections and reconnections. For a mobile application of talKKonnect, you should also engineer a safe shutdown (by button and possibly include a battery monitor script) or you will be risking file system corruptions.

- Upon losing a network connection to Mumble server, talKKonnect dialer won’t be retrying to reconnect to a server forever. It will give up after some time. This is by design. Dial function running forever could result with excessively high CPU utilization, overheating and thermal throttling or a shutdown in some boards. After a predefined number of unsuccessful reconnection attempts, talKKonnect program will terminate. It is the [talKKonnect system service](https://github.com/talkkonnect/talkkonnect/tree/main/conf/systemd) that is responsible for restarting program every … seconds, when a new connection attempt will be made. If still no connectivity, talKKonnect will terminate again, system service will wait and try to launch program again and so on.  

- There is another issue that might occur with sudden network outage. Mumble server may not become immediately aware, that talKKonnect client (or other types of Mumble clients) disconnected. When the client tries to reconnect again, Mumble server may refuse new connection because it thinks the user name is already taken. Server didn’t detect it is the same client trying to reconnect after a network outage. Server may need restart, or client temporarily changing name in order to connect again.

zrdimetc, 2022-01-29:
(2) For connectivity over GPRS/3G/4G data you need to create a script for managing ppp. 

- ppp daemon supports a “persist” option for reconnections. But pppd can be buggy and persist option may not always work.  

- Some 3G modems do not redial successfully after losing a ppp connection, regardless of persist option. You may need apply “brute-force”, unplug them physically and plug again. To work around this, you can use some method of disconnecting and reconnecting USB device without a physical intervention, whether udev rules, USB resetting, binding/unbinding, writing directly to `/sys/bus/usb/` authorized devices. Modem resetting can be included in ppp script. When a script detects an internet outage, modem device should be reconnected to USB bus by a command and ppp dialer redials again and reconnects to APN. You could manage all this through a system service or cron job.  

Talkkonnect is not a toy. It is a powerful “lego-bricks” like framework for PTT communication. You are given complete freedom to choose whether bricks should be mounted one way or another and adjust the final flavor of talKKonnect to your own needs. ppp should be handled at system networking level, not by talKKonnect. When ppp service is properly configured and bug-free, talKKonnect will work happily with it. Please don’t blame talKKonnect for pppd bugs, 3G modem issues or network config issues. With some effort, they can all be successfully overcome in case of your particular modem and talKKonnect can be a very useful companion on the road. Another useful feature when talKKonnect it taken to the road, is it’s [GPS function](https://github.com/talkkonnect/talkkonnect/blob/main/gps.go) and integration with [Traccar](https://www.traccar.org).

zrdimetc, 2022-01-29:
(3) I have had no issues if talKKonnect was tethered over a smartphone or connecting through a mobile hotspot. When phone is back in the coverage area, talKKonnect will reconnect again through phone. There could be reconnection issues with some phones or particular networks, but nothing in talKKonnect to cause issue with reconnection.

BeJay8675309, 2022-01-29:
3G is almost phased out where I live but is still in remote areas, so I'm basing this off 4G. The modem/dongle unit I'm using is really a "full on router", and does the ppp/chap stuff internally, thus is presented to talKKonnect as a USB ethernet adapter (eth0). This problem of it locking up and going AWOL happens when the internet is corrupted AS it's loses 4G. talKKonnect will become unresponsive and no amount of watchdog/script can restart it after this occurs. I'll try and do a video of it for clarity when I have someone else to help film/drive..

zrdimetc, 2022-01-29:
@BeJay8675309
1. Test. Connect to Raspberry Pi over serial UART adapter, establish SSH connection over serial port and use it for control of the unit. Use LAN port for Internet connection. Unplug the cable or bring down eth interface or shut down switch port. Wait for several minutes and re-enable LAN connection. Do you see talKKonnect re-establish connection with your Mumble server?
2. Again, use serial UART for control of the unit. After you lose connectivity over the 4G modem, assuming you get back in the coverage area, remove modem physically and plug again. Is the Internet connection back?
3. What kind of software is used for modem dialing to provider APN?
4. What is you modem model? `lsusb`? `dmesg`? Do you see modem as `/dev/ttyUSB` device?

BeJay8675309, 2022-01-30:
re-1. I was just going to ssh in with my laptop on the same wifi as the pi3 running talKKonnect and run it live from ssh.

re-2. As a side note, I have already tried just pulling the plug on either the usb or ethernet (tried both) and it behaves normally. I can't get it to lock up when the connection is just dropped. It works fine. The issue is when this connection stays up, but is unable to reach the mumble server.

re-3. The software is internal to the 4G modem/router. This dongle also has wifi AP built in.

re-4. i have a HUAWEI 4G Dongle E3131. It shows up as eth0 via usb.

dmesg ->
[  528.385865] usb 1-5: Product: HUAWEI MOBILE
[  528.385866] usb 1-5: Manufacturer: HUAWEI MOBILE
[  528.510840] usb-storage 1-5:1.2: USB Mass Storage device detected
[  528.514552] scsi host6: usb-storage 1-5:1.2
[  528.628407] cdc_ether 1-1:1.0 eth0: register 'cdc_ether' at usb-3f980000.usb-1, CDC Ethernet Device,
[  528.629096] usbcore: registered new interface driver cdc-ether as eth0

lsusb -> 
Bus 001 Device 003: ID 12d1:14db Huawei Technologies Co., Ltd. E353/E3131
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

zrdimetc, 2022-01-30:
@BeJay8675309. You have a Huawei modem working in CDC Ethernet device mode. For managing cellular data connections you can use [NetworkManager](https://wiki.gnome.org/Projects/NetworkManager) and [ModemManager](https://www.freedesktop.org/wiki/Software/ModemManager). You will find them as `network-manager` and `modemmanager` packages in Raspberry Pi OS repositories. Some people prefer to use [usb-modeswitch](https://www.draisberghof.de/usb_modeswitch) to switch from CDC to a modem mode. In modem mode, device usb id will change to` 12d1:1001` and you should see /dev/ttyUSB* devices in the system. Stop by the [usb-modeswitch forum](https://www.draisberghof.de/usb_modeswitch/bb/) and research how the owners of Huawei "HiLink" modems similar to yours, are dealing with modem problems you described?

BeJay8675309, 2022-01-30:
Thanks for the reply zrdimetc, However this issue is really about making talKKonnect more robust to handle this. The Pi itself can recover from the loss and will have internet after the event. My laptop or phone connected to the same modem/network's via wifi, also recovers fine. If I connect to the pi (running talKKonnect) over wifi to the modem, the original issue still occurs negating the issue with the modem itself or the fact it's connected as CDC over USB. Is this a Golang issue with networking?

zrdimetc, 2022-01-30:
@BeJay8675309. After a network outage, all my talKKonnect devices always reconnect successfully to Mumble servers over any network connection or interface, whether it is `eth0`, `wlan0`, 3G modem` ppp0`, `tun/ tap` interfaces, phone tethering, phone Wifi hotspot. Unless there is a server issue! I can not reconstruct issue you describe. How do you start your program instance? Do you use recommended [system service](https://github.com/talkkonnect/talkkonnect/tree/main/conf/systemd)? How many instances of talkkonect do you see running after a network outage? Did you check Mumble server log file for possible refused connections? Is Mumble server seeing any talKKonnect authentication attempts after a network outage? What does `talkkonnect.log` report? You should see program closing and program re-launches and new connection attempts in the log. If your device Internet access is restored, as you say after a brief 4G network outage, but your talKKonnect doesn't reconnect to Mumble server, there could be just 2 reasons. You are not starting program correctly (check system service file) or the network drop out was so sudden that Mumble server didn't detect client disconnection and new connections will be refused by server, because user name "is used". Try a different user name or restart server. Check your Mumble server log! If this is the case, issue has nothing to do with takKKonnect, it is a Mumble server bug. In either case,  issue you describe has nothing to do with talKKonnect "robustness".

BeJay8675309, 2022-01-31:
I'm using the recommended startup system service.

There is only one instance of tk running when this occurs which hangs.

There are no refused connections on the server as it has no internet.

I've got to drive back in the dead zone and test this again with full access to the shell with my laptop, and report back.

zrdimetc, 2022-01-31:
Are you using Raspberry Pi Zero? Is board powering stable 5V? How much current is the modem taking? How high is the CPU utilization and memory load when you are having issue?

BeJay8675309, 2022-01-31:
I've used several Pi's to test this, but the stable rig is a Pi3-A with a "respeaker" shield and a stable 5V supply from 12V using a DC-DC converter
[image]
.

zrdimetc, 2022-01-31:
If GPIO 14 and 15 in Pi are free, connect serial UART adapter to your computer, use for console access for troubleshooting. If pins are taken, consider freeing serial GPIO temporarily. I think no need to drive out for testing network drop out and reconnections yet. Look up Suvir or me on talKKonnect Mumble community server `mumble.talkkonnect.com` and we can go together through a list of actions to try to remedy your modem issue. No promises. But it will be easier from typing back and forth.

talkkonnect, 2022-02-01:
BeJay8675309,
something comes to mind, can you please try the following xml tag from <hardware targetboard="rpi"> <hardware targetboard="pc"> and then connect to your bad signal cellular network.

My suspicion is that the GPIO polling I wrote may be behaving badly, this will disable GPIO from working just to experiment if this causing the lockup. Also please see the CPU level when it locks up does the CPU go to 100%?

Let us know, as Zoran said earlier We are on the channel mumble.talkkonnect.com port 64738 with password talkkonnect, maybe if we talk using voice this issue will be easier to fix without the communication delay.

rotary encoder and max7219 seven segment channel control is now working inside talkkonnect, no need external api calls for all this.

happy talkkonnecting!

BeJay8675309, 2022-02-01:
Will re-adjust my hardware to expose the serial port and give it a go in the next few days. Thanks for your time looking into this.

zrdimetc, 2022-02-01:
Hi Suvir @talkkonnect. Yes, it is always a good troubleshooting step to change mode from <hardware targetboard="rpi"> to `<hardware targetboard="PC">` in `talkkonnect.xml` as you recommend and run program without GPIO functions and no display. It's one of the many great features in your design. @BeJay8675309 should monitor program from serial UART, not a network interface. After a network outage, `talkkonnect `will try to reconnect `3 times` to Mumble server, then terminate. System service will wait and re-launch talkkonnect after 10 seconds when new connection attempts will be made... and so on, until a connection to Mumble server can be re-established. User can see `screen` and `talkkonnect` processes starting and closing by watching with `top `or `htop`. Recommended practice for modem management in 2022 and current state of the affairs with modern Linux distributions such as `Raspbian Pi OS`/ `Debian 10` or `11` is to use `Network Manager` and `Modem Manager` packages and let them handle network connections and reconnections automatically. But of course, users are free to choose whatever method for managing modem and networks as they feel confident with. `Network Manager` is working for me with for 'gsm' type connections over ttyUSB* modem devices. It is "plug-and-play", no issues, no additional network scripting required. However, @BeJay8675309 is using a "HiLink" modem which acts like `USB Ethernet gadget` or `RNDIS` `g_ether` / `cdc_ether` device. It is also possible to set his modem to `tty` mode. Provided the `Network Manager` is used, this type of connection clearly needs to be set as "`Ethernet`" (with interface `MAC` address). Some of these modems act as routers with `NAT` and will use address like `192.168.8.1`. There is a WEB GUI for managing the modem. Modem/router should have a DHCP server capability or another DHCP server can be used. The user device should get IP address in the same `/24` net range. Or `eth 192.168.8.X/24` address with` gw 192.168.8.1` can be set statically. There used to be all kinds of issues with Huawei `HiLink` modems and Linux  and `Network Manager`. Modem bugs are not a "talKKonnect issue". Users should select modem models carefully, based on price and working with OS without issues. This may require testing with trials and errors, before narrowing down choice to a modem model. Its hard to give modem recommendations, situation is different in every country with networks and modems availability. I would say, don't go for a modem hat that "breaks the bank", just get a simple affordable modem that works for you. With talKKonnect devices, I use old 3G modems like Huawei HDSPA E1550, E1552, E1752, E153 in tty/gsm mode. Some countries still operate 2G/2.5G... GPRS and 3G and have them coexist with 4G/LTE. Situation differs from country to county and this will influence a modem selection.

zrdimetc, 2022-02-01:
However, @BeJay8675309 is using a "HiLink" modem which acts like `USB Ethernet gadget` or `RNDIS` `g_ether` / `cdc_ether` device. It is also possible to set his modem to `tty` mode. Provided the `Network Manager` is used, this type of connection clearly needs to be set as "`Ethernet`" (with interface `MAC` address). Some of these modems act as routers with `NAT` and will use address like `192.168.8.1`. There is a WEB GUI for managing the modem. Modem/router should have a DHCP server capability or another DHCP server can be used. The user device should get IP address in the same `/24` net range. Or `eth 192.168.8.X/24` address with` gw 192.168.8.1` can be set statically. There used to be all kinds of issues with Huawei `HiLink` modems and Linux  and `Network Manager`. Modem bugs are not a "talKKonnect issue". Users should select modem models carefully, based on price and working with OS without issues. This may require testing with trials and errors, before narrowing down choice to a modem model. Its hard to give modem recommendations, situation is different in every country with networks and modems availability. I would say, don't go for a modem hat that "breaks the bank", just get a simple affordable modem that works for you. With talKKonnect devices, I use old 3G modems like Huawei HDSPA E1550, E1552, E1752, E153 in tty/gsm mode. Some countries still operate 2G/2.5G... GPRS and 3G and have them coexist with 4G/LTE. Situation differs from country to county and this will influence a modem selection.

BeJay8675309, 2022-02-02:
Lets put this "modem issue theory" aside for a moment... After activating the serial uart on the pi, I have so far not had any problems. Weird? Normally it will go 100% cpu and hang, but so far I have not been able to replicate the issue with the console enabled.

When TXing and network drops:
AL lib: (EE) ALCcaptureAlsa_captureSamples: read error: Resource temporarily unavailable
AL lib: (EE) ALCcaptureAlsa_captureSamples: read error: Broken pipe
AL lib: (EE) ALCplaybackAlsa_mixerProc: mmap commit error: Broken pipe
[ debug ] 2022/02/02 17:24:31 gpio.go:392: Turning Off All LEDS!
[ alert ] 2022/02/02 17:24:31 onevent.go:109: Attempting Reconnect in 10 seconds

and it resumes normally. 

When RXing and network drops:
AL lib: (EE) ALCcaptureAlsa_availableSamples: avail update failed: Broken pipe
[ debug ] 2022/02/02 17:40:21 gpio.go:392: Turning Off All LEDS!
[ alert ] 2022/02/02 17:40:21 onevent.go:109: Attempting Reconnect in 10 seconds

and it resumes normally.

talkkonnect, 2022-02-02:
So the problem gets magically solved if serial enabled?

BeJay8675309, 2022-02-02:
It seems that way so far, will test some more as I have another identical rig I built without serial enabled.

talkkonnect, 2022-02-02:
BeJay thanks for your feedback and testing much appreciated. Do let us know the results of your other rig.

zrdimetc, 2022-02-02:
@BeJay8675309. Probably your `ssh `session over `eth0 ` was getting frozen earlier? Was your `eth0` network config `static `or `dhcp`? Your routing or `dns `entries in `resolv.conf` change with 4G network disconnection. It was not `talkkonnect` program freezing earlier, it must have been your ssh session. Have fun with talKKonnect on the road.

zrdimetc, 2022-02-02:
@BeJay8675309. You have OpenAL errors above "`AL lib: (EE) ALCplaybackAlsa...`", as if your sound device was not recognized or had lost connection to the board. OpenAl configuration is in `/etc/openal/alsoft.conf`, and usually default settings work, no changes necessary for talkkonnect, except for anyone eager to experiment. Regards.

BeJay8675309, 2022-02-02:
@zrdimetc it wasn't my ssh session freezing, as the issue would occur without shell interaction, as in just running as a service directly on the metal (inside screen). The openAL errors are another issue with the re-speaker hat I'm using that isn't re-initialized properly after talKKonnect is stopped during capture. This doesn't occur with a USB audio interface so we can chalk that up to a bad hardware driver/implementation i think. I just restart alsa-state service and the delayed audio issue is fixed. 

Will test the uart on and off with both rigs tomorrow and see if enabling the uart is all it takes to fix the hanging issue...

zrdimetc, 2022-02-03:
@BeJay8675309. Serial enabled or not, has nothing to do with stability or performance of `talkkonnect `program. It's not a requirement. If configured properly, current version of talKKonnect will run stable. System script working together with program net dialer will handle connections and reconnections. Since you referred to program locking up or freezing in relation to modem losing net connection, debug over serial was suggested as a way of troubleshooting. You reported an issue with your custom build running your own `python `GPIO script, but provided no logs, messages or errors. I was not able to reproduce issue with any of my builds in spite of the effort. This doesn't mean other users won't come across issues with different modems or network configs, but these are exclusively OS issues outside of talKKonnect. Pehaps some of the comments will be of help to them. Good luck fixing your build and hopefully your rigs will perform up to your expectations. Best regards.

BeJay8675309, 2022-02-04:
It took a while, but it still locks up. The logs don't show anything useful, that's why I haven't included them. I recompiled talkkonnect from scratch running stock. No python api interface, and it will still lockup. The screen console becomes unresponsive and ssh'ing in I can see top reporting 100% cpu.

   PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                          
  663 talkkonnect  20   0  921296  31032  15652 S   100.0   7.1 127:53.74 talkkonnect    

It seems harder to reproduce yesterday, but it happened on both rigs at different times. 

I will continue to test different hardware and configurations to see if I can eliminate this. 

Thanks for your time.

zrdimetc, 2022-02-04:
@BeJay8675309. I would really like to help you, since you put a lot of effort in designing your rig, but I am running out of ideas. Try some of the talKKonnect images Suvir had posted as a base line? Always test first in `pc` mode, and enable `rpi` mode later. Can you confirm your issue if occurring in pc mode too? Try a different modem or change E353 working mode from "HiLink" `cdc_ether` to "Broadband" `ttyUSB` by using `usb_modeswich` utility? Try a Pi 3B+ board. Although it features the same CPU and memory is not an issue for talKKonnect. My 3B+ builds work stable with modems in `tty `mode, no 100% CPU hogging you experienced.

BeJay8675309, 2022-02-04:
Thanks for your ideas. I will continue to do some experiments with different boards etc when I have the time. I will also run it on a vm (x64) on my laptop with the same modem and see if the issue still occurs.  Feel free to close this ticket and I'll reopen it if I find anything useful. Thanks again for your persistence. It's great to have helpful devs.

closed 2022-02-04 by talkkonnect

zrdimetc, 2022-02-05:
If you want to build talKKonnect on a PC/ laptop, you will need to use old [gopus](https://github.com/talkkonnect/gopus) before Suvir updating Opus. Use the initial commit.

Keksstar, 2024-02-26:
@zrdimetc Two years later... I have a question on this. Config is as follows:
Router -EthernetCable-> Raspberry
I´ve set an fixed IP on the Raspberry and also Gateway/DNS.

Now I´m wondering:
When I boot up the Raspberry and connect the ethernet afterwards: I got no connection. I only get an connection when I´m rebooting the Raspberry. I get an instant connection when the ethernet is connected from the beginning.

I´m in a scenario, where I power everything by battery. But the Raspberry has less boottime then the Router, so I´m wondering why I don´t get an connection? As you wrote above, Talkkonnect is trying to establish an connection three times, then reboots and trys again...