talkkonnect/talkkonnect #46: WIFI doesn't seem to function on pi 4 w/ this image.

bikr, 2021-08-17:
Tried multiple "fixes" to enable WPA Supplicant etc...  The onboard wifi of the raspberry pi 4 simply does not seem to work.

talkkonnect, 2021-08-18:
The images were created using a raspberry 3b+ unfortunately I don’t have a 4 but the source install works for the raspberry pi 4 perhaps there is a different driver or some other incompatibility between the 2 versions.

zrdimetc, 2021-08-19:
You are not providing more specific problem description, but we can assume it is Raspberry Pi 4 wireless bug. You can confirm this by researching issues already raised at different other forums dealing with RPi4. There was a documented issue about RPi4 wifi misbehaving when using with HDMI high resolution. Some possible solutions could be... 1) upgrading your wifi firmware (`firmware-brcm80211`). 2) dont use high resolution HDMI. 3) Make sure you set your timezone/location for RPI4... 4) Ensure your power supply is adequate. 5) Check your wpa supplicant.conf for correct settings and study wpa_supplicant log file for errors. After you solve the wifi problem, there is no reason why the talKKonnect build wont work on your device, provided your configuration in talkkonnect.xml is correct. If you still encounter problems, try building from source code on a fresh Raspbian OS.

zrdimetc, 2021-08-24:
Suvir's talKKonnect [image](https://drive.google.com/file/d/1hbMFtKvlEYX-akqf976aVjHP4TcYFXgL/view?usp=sharing) was created for RPi 3. If you intend to run it on RPi 4, we recommend the following. Provide wired network connection to RPi 4 and run `apt-get update` and `apt-get upgrade`. Check that upgrade completes and you get the latest RPi firmware .
Enter wifi SSID and password through raspi-config or by editing `/etc/wpa_supplicant/wpa_supplicant.conf`.

```
root@raspberrypi:~# ifconfig wlan0 up
SIOCSIFFLAGS: Operation not possible due to RF-kill
```

```
root@raspberrypi:~# rfkill list
0: phy0: Wireless LAN
        Soft blocked: yes
        Hard blocked: no
1: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
```

Unblock phy0 with

`root@raspberrypi:~# rfkill unblock 0`

Attempt to bring wlan0 up will be successful.

`root@raspberrypi:~# ifconfig wlan0 up`

Proceed with your talKKonnect configuration and adjustments as desired.

Regards,

talkkonnect, 2021-08-24:
Problem solved closing ticket

closed 2021-08-24 by talkkonnect