talkkonnect/talkkonnect #47: I2C Interface not working with values from Sample Config

moffi9426, 2021-08-19:
This is most certainly an issue on my side, but I cannot seem to track it down. As soon as I try to enable I2C Talkkonnect doesn't start with an error code. If the interface is set to false it works like a charm:

```
panic: non-positive interval for NewTicker

goroutine 1 [running]:
time.NewTicker(0x0, 0x0, 0x1)
	/usr/local/go/src/time/tick.go:23 +0x168
github.com/talkkonnect/talkkonnect.(*Talkkonnect).ClientStart(0x1306000)
	/home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/client.go:310 +0x46c
github.com/talkkonnect/talkkonnect.Init(0x64264f, 0x4f, 0xbe982819, 0x1)
	/home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/client.go:251 +0x66c
main.main()
	/home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/cmd/talkkonnect/main.go:78 +0x148

```

From the XML (i2C adress is 0x27, targetboard is set to "rpi"):
```
                       <lcd enabled="true">
                                <lcdinterfacetype>i2c</lcdinterfacetype>
                                <lcdi2caddress>39</lcdi2caddress>
                                <lcdbacklighttimerenabled>false</lcdbacklighttimerenabled>
                                <lcdbacklighttimeoutsecs>30</lcdbacklighttimeoutsecs>
                                <lcdbacklightpin></lcdbacklightpin>
                                <lcdrspin></lcdrspin>
                                <lcdepin></lcdepin>
                                <lcdd4pin></lcdd4pin>
                                <lcdd5pin></lcdd5pin>
                                <lcdd6pin></lcdd6pin>
                                <lcdd7pin></lcdd7pin>
                        </lcd>

```

zrdimetc, 2021-08-20:
Try to build with a code before August 19 commits. There is a code clean up and integration of voice targeting and keyboard features going on. Problem you report is caused by `LCDBackLightTimeoutSecs` variable name changing to `LCDBackLightTimeout`. Keep checking the new versions. Issue will be fixed soon by Suvir, talKKonnect author. Or you can adjust the "LCD backlight timeout" variable references in your current code, so that name matches. You have to edit `xmlparser.go`, `stream.go` and `talkkonnect.xml`, find or grep this variable. Follow a well written and documented code from Suvir and use debug feature in the console screen and you should fix this easily. For using the new voice targeting and keyboard features, the xml config needs upgrading, Keep checking for the new config version, when it gets releases and the news from Suvir.

talkkonnect, 2021-08-20:
I broke the LCD function unintentionally I have attempted a fix in the latest release. Will be tested and inform you again. As Zrdimetc said in the post above it has to do with the LCD backlight timer variable and not being able to pick this setting from the xml file.

moffi9426, 2021-08-20:
I saw you pushed a new commit, current situation: As soon I enable LCD talkkonnect fails to unmute my device and I get the following error in the logging:

`[ error ] 2021/08/20 16:39:54 client.go:581: Key Not Mapped ASC  0`

If I disable LCD unmuting works fine and I can connect.

zrdimetc, 2021-08-20:
Please provide more details about your sound device and LCD type? Did you detect display i2c hex address with `i2cdetect` as `0x27` hex and then converted this to decimal `39`? Arduino use `7-bit` addresses and for RPi we need `8-bit`. If the LCD display was designed and marketed as Arduino part, ignore `7-bit` address, if printed on the back, detect it as 8-bit with `i2cdetect` tool from  `i2c-tools` package. Ascii code "`0`" should be a space key. The latest keyboard changes are not yet enabled in the current `talkkonnect.xml` and do not influence your issue. Most HD44780 LCD's with PCF8754 i2c I/O expander, we have come across,  were preset for `0x3F` / decimal `63`.

moffi9426, 2021-08-20:
Yes I detected the hex address with `i2cdetect -y 1` correctly. I can confirm it works when loading a normal raspbian and throw a testprogram at it, displays everything just fine. However the I2C adapter just has a jumper to control backlight, but that's fine, since always on backlight is fine.

Sound device is a wireless headset from Amazon with a USB dongle, shows up in alsamixer (of course I copied the name from alsamixer to the `.xml` just fine as intended and as long I don't set the LCD to "true" it works as it should

zrdimetc, 2021-08-20:
Try to disable LCD back light timer in the config. Backlight timer works differently for i2c and parallel display. There is a pin controlling LCD led through a transistor in parallel LCD design. This allows dimming the led to the desired low intensity instead of turning led illumination completely off. Disable LCD backlight timer in your current tests. Also, temporarily remove headset from equation. while troubleshooting. Use RPi BCM audio, it should work for receiving and while testing the display for your issue. If talking without audio input device, talKKonnect may crash or use any other USB sound card with audio input available. Does the headset have media keys or some controller that initializes and acts like a keyboard?

moffi9426, 2021-08-21:
Got the unmute fixed, that was an issue on my side, I had a typo in the name of the audio device. The Headset only got a mute button apart from on/off, but that seems to be headset internal.

If I now start with LCD enabled I get the following error:

```
panic: embd: your host ": ARMv7 Processor rev 3 (v7l)" is not supported at this moment. request support at https://github.com/kidoman/embd/issues

goroutine 50 [running]:
github.com/talkkonnect/go-hd44780.(*I2C4bit).Open(0x1044f78, 0x1ef70, 0x11eb870)
	/home/talkkonnect/gocode/pkg/mod/github.com/talkkonnect/go-hd44780@v0.0.0-20190204103905-60500cf76000/i2c_conn.go:55 +0x1fc
github.com/talkkonnect/go-hd44780.LcdDisplay(0x6123de, 0x3, 0x6123de, 0x3, 0x6123de, 0x3, 0x6123de, 0x3, 0x0, 0x0, ...)
	/home/talkkonnect/gocode/pkg/mod/github.com/talkkonnect/go-hd44780@v0.0.0-20190204103905-60500cf76000/LcdDisplay.go:54 +0x2c0
created by github.com/talkkonnect/talkkonnect.LcdDisplay
	/home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/display.go:88 +0x90
```

zrdimetc, 2021-08-21:
Raspberry Pi 3B+ was most adequate hardware for running talKKonnect's when the project started and probably still is. talKKonnect code was developed and tested by Suvir largely focusing on this particular board. `kidoman/embd` library used for i2c support for displays was created long before Raspberry Pi 4. There could be other issues with other libraries preceding the launch of RPi4, used in talKKonnect, embd created long before RPi4 is unaware of armv7l (small "L" not "1") and objects to being called from RPi4. If you are determined to making your build work with i2c LCD and RPi4, try to tweak [detect.go](https://github.com/kidoman/embd/blob/master/detect.go) to make embd understand it is now working with RPi4, or try to bypass the hardware check altogether. Keep checking for updates from Suvir, embd issue may get fixed later on.

zrdimetc, 2021-08-21:
Try to edit `detect`.go from `kidoman/embd`
to add ARMv7l and BCM2711.
```
	switch {
	case strings.Contains(model, "ARMv8" || "ARMv7l" || "ARMv7") && (strings.Contains(hardware, "AM33XX") || strings.Contains(hardware, "AM335X")):
		return HostBBB, rev, nil
	case strings.Contains(hardware, "BCM2711") || strings.Contains(hardware, "BCM2709") || strings.Contains(hardware, "BCM2708") || strings.Contains(hardware, "BCM2835" || strings.Contains(hardware, "BCM2836" || strings.Contains(hardware, "BCM2837" || strings.Contains(hardware, "BCM2837A0" || strings.Contains(hardware, "BCM2837B0"):
		return HostRPi, rev, nil
```
Let us know.

moffi9426, 2021-08-22:
I'm not too familiar with go, so maybe somebody can point me to the error. I added the changes your mentioned already yesterday followed by

`go build -o /home/talkkonnect/bin/talkkonnect cmd/talkkonnect/main.go` with root privileges

Also for checking if compile went okay I slightly altered the error message. When I start from `/home/talkkonnect/bin` I still get the old (unmodified) error message. Of course I'm cleaning up the directories before compiling.

zrdimetc, 2021-08-22:
I wired HD44780 20x4 display and PCF8754 i2c expander with RPi 4B and replicated the issue you came across. `kidoman/ embd` is a requirement for `go-hd44780` and i2c. Unfortunately, library is quite old, dates back to RPi 2 time. Until a solution can be found, try HD44780 in parallel mode? Oled SSD1306 1.3 inch i2c should also work.

zrdimetc, 2021-08-22:
I can confirm that patching `detect.go` in `kidoman/embd` works to successfully compile talkkonnect with working i2c support for HD44780 LCD display. It is sufficient to extend case statement with Broadcom BCM2711 CPU.
` 	case strings.Contains(hardware, "BCM2708") || strings.Contains(hardware, "BCM2709") || strings.Contains(hardware, "BCM2835") **|| strings.Contains(hardware, "BCM2711")**:`
A [pull request](https://github.com/kidoman/embd/pull/98) for this was made a year ago, but never committed to `embd`. Suvir will probably fork `kidoman/embd` and then the issue will be fixed for talkkonnect builders.  Applying changes directly in ` /home/talkkonnect/gocode/src/github.com/kidoman/embd` wont work because `go build` will be picking code from `kidoman/embd inside /home/talkkonnect/gocode/pkg/mod/github.com/kidoman/embd@v0.0.0-20170508013040-d3d8c0c5c68d`. As a temporary fix, please navigate there and apply the above change manually to include BCM2711, then recompile and run talkkonnect again. Display should work.

talkkonnect, 2021-08-23:
Ok I will fork and patch kidoman and put it under the talkkonnect umbrella.

talkkonnect, 2021-08-23:
I have cloned and added zrdimetc  recommended patches. The library is now vendored and available in https://github.com/talkkonnect/embed with these changes hopefully a clean install can now work with I2C on raspberry 4. Please test and let me know.
Thanks again for all your testing and feedback.

zrdimetc, 2021-08-23:
Many thanks Suvir. [Broadcom SoC descriptors for Raspberry Pi's](https://www.raspberrypi.org/documentation/computers/processors.html). We can expect to find the following SoC,

BCM2835: in RPi A, B, A+, B+, Zero, Zero W, Zero WH, CM
BCM2836: in RPi 2B
BCM2837 (BCM2837A0): in RPi 2B v1.2 and RPi 3B
BCM2837 (BCM2837B0): in RPi 3A+, 3B+
BCM2711: in RPi 4B, RPi 400

I will test with 2's and 3's later today and you can probably then close the issue.

moffi9426, 2021-08-23:
Can also confirm it works as per zrdimetc changes made in "pkg". Display works now. 

Last issue I'm still having. Even though I've got the backlighttimer set to `false` in the xml I still get `[ debug ] 2021/08/23 16:24:15 client.go:315: LCD Backlight Ticker Timed Out After 30 Seconds ` after the given timeout in the xml and screen doesn't show anything anymore until I restart.

zrdimetc, 2021-08-23:
Display should wake up when some talKKonnect function calls HD44780, for instance you press a PTT, some button or there is activity in the channel. It's how backlight control works with i2c. Backlight led is on or off, no in between. If this is undesired behavior you could turn the backlight timeout function off, just dont use it. Or make the timeout much longer. This is why we prefer 8-bit parallel mode with this LCD. You can control backlight led with a transistor, add a trimer pot and you can adjust low intensity illumination for backlight led any way you like and display will light up again in full intensity on an event.

moffi9426, 2021-08-23:
Still there's in issue in the `xmlparser.go`. Even though I disable the timer the debug on startup says it's enabled.

Line 1488 needs to be changed from 

`LCDBackLightTimerEnabled = Document.Global.Hardware.LCD.Enabled` 

(this pulls the value for the LCD being enabled/ disabled at all) to 

`LCDBackLightTimerEnabled = Document.Global.Hardware.LCD.BacklightTimerEnabled` (see line 754).

While I was at it I renamed `BacklightTimerEnabled` (also small l instead of capital l) to `LCDBackLightTimerEnabled` to keep the code clearly documented.

This resolves all my issues with the screen. Thanks @zrdimetc @talkkonnect for your great help and feedback. In my opinion we can close the issue now.

zrdimetc, 2021-08-23:
Glad if we could help. Hope you will like talKKonnect and find a good practical use of it in your setting. Suvir @talkkonnect  made a great framework for building PTT communication system, adding a lot of innovative practical functions on top of great work of [Gumble](https://github.com/layeh/gumble) and [talkiepi](https://github.com/dchote/talkiepi) authors. At the same time, features can be set at a very granular level, enabled or disabled, adjusted to a particular need through xml configuration. Hardware is easily found in most countries and doesn't cost "arm and leg". talKKonnect is like a "LEGO bricks" kit for building a PTT system. Best regards.

talkkonnect, 2021-08-24:
Problem solved closing ticket

closed 2021-08-24 by talkkonnect