talkkonnect/talkkonnect #48: how to listen mesages that are writen from android or windows chat window

spereg53, 2021-08-19:
i have connected with several talkkonnect and interconnected to a radio 2ways type ,what can i do for listen when someone wite a text in chat mode???,thanks

talkkonnect, 2021-08-21:
The latest version of talkkonnect 1.67.11 Released Aug 21 2021now  supports this requested feature. To enable or test this, 
 in talkkonnect.xml under the software xml tag you can add the following and it should read back the message to you and also key the transmitter if you want based on the localplaywithrxled tag setting

                   <ttsmessages enabled = "true">
                                <localplay>true</localplay>
                                <localplaywithrxled>true</localplaywithrxled>
                                <playintostream>false</playintostream>
                        </ttsmessages>

spereg53, 2021-08-21:
please ,can you give the link ,where can i download the new image???,thanks
in advance
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

talkkonnect, 2021-08-21:
No image yet, use the script to update it. Get the latest script update-talkkonnect.sh from github.com in the scripts directory and put it in the /root directory. Cd to the root directory and run it. Then you should have latest version.

spereg53, 2021-08-22:
i did what you wrote but the file talkkonnect.xml,does not change ,because
copy the old file ,the version change,but not the file:talkkonnect.xml,what
can i do?????,thanks
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

spereg53, 2021-08-22:
make a clean installation and is upgraded and change the parameters of tts
-message,but i can not get audio on the raspberry,when chat on my pc,thanks
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

talkkonnect, 2021-08-23:
You will need to install ffplay for the audio to work. Try to install ffplay.

spereg53, 2021-08-23:
was lookinf for ffplay and mus be install ffmpeg ,i did it ,but did not
work,any ideas???thanks
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

talkkonnect, 2021-08-24:
Try to play any mp3 file with ffplay direct to speaker using the command directly from the bash command prompt without anything to do with talkkonnect, does it work and can you hear the mp3 file playing normally?

spereg53, 2021-08-24:
Ok, I will try.
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

spereg53, 2021-08-24:
yes a i did with the command "ffplay test.mp3" an file that was record with
audacity and is ok sound like i recorded,now what i have to do for work
with ttsmessage????thanks
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

zrdimetc, 2021-08-24:
Just build program from the latest code and enable text message to TTS playing option with 
```
<ttsmessages enabled = "true">
<localplay>true</localplay>
<localplaywithrxled>true</localplaywithrxled>
<playintostream>true</playintostream>
</ttsmessages>
```
Only a single device should have this enabled, Then it will play TTS to the speaker, channel or both.

spereg53, 2021-08-24:
not woringk, and when i talk with other client ,the voice  of that client
get a lot of time to play on tlakkonnect installed on the raspberry if
takeout the ttsmessage is normal the operation of mumble system ,there in
not delay,what should i do???,thank for helpme
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

zrdimetc, 2021-08-25:
I think this is by the intended design of new TTS playing function. You can choose playing TTS locally to the speaker. And you can use this for interfacing with other device like radio. Or you can play TTS on the Mumble channel to users. You can enable both,  but TTS playing will be sequential, first local to speaker, then to the channel. If this is undesired, use just one of the TTS playing modes `<localplay>` or `<playintostream>`, but not both.

spereg53, 2021-08-25:
but does not work neither in local play neither channel,what i am doing
wrong???thanks in advance
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

talkkonnect, 2021-08-25:
The tts engine uses google dies the raspberry pi have internet access?
This is the final answer we have we are out of ideas why this tts function
doesn’t work for you
-- 
Suvir Kumar
Global Coders Co., Ltd.
55/1 Bandon-Cherngtalay road, Cherngtalay, Thalang, Phuket 83110
Tel. 668 1 840 7160

spereg53, 2021-08-25:
No, it doesn't. thanks anyway
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

talkkonnect, 2021-08-25:
Ok problem solved you need internet access for the text to speech to work
since it relies on google cloud. That’s why it’s not working for you
because your device has no internet access.
-- 
Suvir Kumar
Global Coders Co., Ltd.
55/1 Bandon-Cherngtalay road, Cherngtalay, Thalang, Phuket 83110
Tel. 668 1 840 7160

spereg53, 2021-08-25:
Thanks, I'll check it out.
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***