talkkonnect/talkkonnect #50: Intermittent high delay when transmitting

maverick077, 2021-08-28:
When I try to autostart talkkonnect on raspberry pi, I refer to README first but it seems no work.
So I directly add: `screen -dmS tk /home/pi/my_path_tk/talkkonnect &` in `/etc/rc.local`, and it works.

However, a new problem is troubling me.
It seems that: in a few minutes, talkkonnect has a high delay(1 to 2 seconds) when transmitting(audio input using microphone); in another few minutes, it is normal; and these two situations alternate. At the same time, audio output always works well.
I don't know where the problem lies, talkkonnect, mumble-server, raspberry pi, or microphone?

Some hints needed, and sorry to bother you.

talkkonnect, 2021-08-28:
If the audio is not choppy and delayed from time to time but still arrives without chopping at the client side this is normal for any mumble client. Using mumble for PTT communications this does not pose a problem, however if you listen to your transmission on another device in the same room you will find this phenomenon. This should be true for any type of voice over IP system too and should not affect or degrade the experience for PTT communications. Are you trying to use it for something that requires a very near real time use case? Something that has many devices in the same vicinity that the sound overlaps each other? Please elaborate.

zrdimetc, 2021-08-28:
@maverick077.  Methods for starting talKKonnect are documented in Suvir's blog [here](https://talkkonnect.com/2020/06/15/various-methods-for-starting-talkkonnect/). Do you use RPi 4B? Do you use wireless or wired LAN? Is it headless or desktop? If desktop, is `pulseaudio` running? How high is the CPU utilization and memory usage on TX? Is CPU temperature normal? Do you use HDMI in high resolution (_there is an issue with RPI 4B wifi and HDMI_)? Where is your Mumble server located, ping response times, any jitter, packet loss? There could be many reasons for what you experience and none would have much to do with talKKonnect / gumble under normal circumstances. Try building with Raspberry Pi 3B, 3B+, 3A+ or 2B and compare? Feel free to join talKKonnect Mumble community server [mumble.talkkonnect.com:64738](url), it may be more quick to go through a check list and narrow down possible causes. They are likely network, OS/ sound sub-system related or RPI 4B hardware specific issue.

maverick077, 2021-08-29:
> If the audio is not choppy and delayed from time to time but still arrives without chopping at the client side this is normal for any mumble client. Using mumble for PTT communications this does not [...]

Yes, I used the laptop and the Raspberry PI 4 on the same LAN and tested them in the same room as you said. I'm glad this is a normal phenomenon and not a setup error or compatibility issue. Thank you for the satisfying answer.

maverick077, 2021-08-29:
> @maverick077. Methods for starting talKKonnect are documented in Suvir's blog [here](https://talkkonnect.com/2020/06/15/various-methods-for-starting-talkkonnect/). Do you use RPi 4B? Do you use wireless or wired LAN? Is it headless or desktop? If desktop, is running? How high is the CPU utilization and memory usage on TX? Is CPU temperature normal? Do you use HDMI in high resolution (_there is an issue with RPI 4B wifi and HDMI_)? Where is your Mumble server located, ping response times, any jitter, packet loss? There could be many reasons for what you experience and none would have much to do with talKKonnect / gumble under normal circumstances. Try building with Raspberry Pi 3B, 3B+, 3A+ or 2B and compare? Feel free to join talKKonnect Mumble community server **mumble.talkkonnect.com:64738**, it may be more quick to go through a check list and narrow down possible causes. They are likely network, OS/ sound sub-system related or RPI 4B hardware specific issues.`pulseaudio`

I used raspberry Pi on wireless LAN without desktop, I think the temperature was normal. Also, I think that the owner's reply is satisfactory to me because my application scenario is exactly what he describes.
Of course, thank you for your detailed hints, which give me a lot of directions to think about.
In addition, does talkkonnect support PulseAudio? I used an older version of the image which only used ALSA to avoid complex compatibility issues. Now, I want to try the latest mirror image of raspberry Pi.

talkkonnect, 2021-08-30:
Pulse audio works but is not for the faint of heart. On the other hand also works out of the box. Depending on your situation you can use either, but I personally prefer alsa it’s much easier.

closed 2021-08-31 by talkkonnect