talkkonnect/talkkonnect #51: I have issue with GPS

haocomm, 2021-09-04:
I use a talkkonnect is very nice project. and Now I would like to use USB GPS feature.

and I have some issue. I'm use USB u-blox7 (VK-172) with raspberry pi 3 B+.
and I'm try to used USB GPS module and I have some error code from logging

(warn) 2021/09/04 06:09:08 scanncer Function Error nmea: sentence prefix 'GPTXT' not supported
(warn) 2021/09/04 06:09:08 scanncer Function Error nmea: sentence prefix 'GPTXT' not supported
(warn) 2021/09/04 06:09:08 scanncer Function Error nmea: sentence prefix 'GPTXT' not supported
(warn) 2021/09/04 06:09:08 scanncer Function Error nmea: sentence prefix 'GPTXT' not supported

then the talkkonnect service is restart . 

I check USB u-blox7 is work fine.
 
tail -f /dev/ttyACM0

$GPRMC,061231.00,A,1759.58434,N,09901.71834,E,0.156,,040921,,,A*77
$GPVTG,,T,,M,0.156,N,0.288,K,A*23
$GPGGA,061231.00,1759.58434,N,09901.71834,E,1,07,0.96,577.7,M,-37.1,M,,*75
$GPGSA,A,3,15,29,05,02,23,24,25,,,,,,1.76,0.96,1.48*08
$GPGSV,4,1,15,02,21,119,37,05,28,041,15,10,05,270,,11,12,116,*7E
$GPGSV,4,2,15,12,04,179,14,13,40,039,,15,70,017,17,18,31,331,17*75
$GPGSV,4,3,15,20,09,067,,23,26,294,21,24,32,153,35,25,07,207,13*78
$GPGSV,4,4,15,29,64,242,41,39,07,265,36,50,52,120,*4D
$GPGLL,1759.58434,N,09901.71834,E,061231.00,A,A*62
$GPTXT,01,01,02,u-blox ag - www.u-blox.com*50
$GPTXT,01,01,02,HW  UBX-G70xx   00070000 *77
$GPTXT,01,01,02,ROM CORE 1.00 (59842) Jun 27 2012 17:43:52*59
$GPTXT,01,01,02,PROTVER 14.00*1E
$GPTXT,01,01,02,ANTSUPERV=AC SD PDoS SR*20
$GPTXT,01,01,02,ANTSTATUS=OK*3B
$GPTXT,01,01,02,LLC FFFFFFFF-FFFFFFFD-FFFFFFFF-FFFFFFFF-FFFFFFF9*53
$GPRMC,061232.00,A,1759.58430,N,09901.71857,E,0.182,,040921,,,A*7C
$GPVTG,,T,,M,0.182,N,0.337,K,A*2F
$GPGGA,061232.00,1759.58430,N,09901.71857,E,1,07,0.96,578.6,M,-37.1,M,,*79
$GPGSA,A,3,15,29,05,02,23,24,25,,,,,,1.76,0.96,1.48*08
$GPGSV,4,1,15,02,21,119,37,05,28,041,16,10,05,270,,11,12,116,*7D
$GPGSV,4,2,15,12,04,179,14,13,40,039,,15,70,017,17,18,31,331,18*7A
$GPGSV,4,3,15,20,09,067,,23,26,294,21,24,32,153,35,25,07,207,09*73
$GPGSV,4,4,15,29,64,242,41,39,07,265,36,50,52,120,*4D
$GPGLL,1759.58430,N,09901.71857,E,061232.00,A,A*60
$GPRMC,061233.00,A,1759.58426,N,09901.71878,E,0.117,,040921,,,A*7B
$GPVTG,,T,,M,0.117,N,0.217,K,A*20
$GPGGA,061233.00,1759.58426,N,09901.71878,E,1,07,0.96,579.5,M,-37.1,M,,*70
$GPGSA,A,3,15,29,05,02,23,24,25,,,,,,1.76,0.96,1.48*08
$GPGSV,4,1,15,02,21,119,36,05,28,041,17,10,05,270,,11,12,116,*7D
$GPGSV,4,2,15,12,04,179,13,13,40,039,05,15,70,017,17,18,31,331,17*77
$GPGSV,4,3,15,20,09,067,,23,26,294,20,24,32,153,35,25,07,207,09*72
$GPGSV,4,4,15,29,64,242,41,39,07,265,36,50,52,120,*4D
$GPGLL,1759.58426,N,09901.71878,E,061233.00,A,A*6B

Can you recommend?

talkkonnect, 2021-09-04:
OK we didnt check for some time, Zoran will help to test and let me know what to fix. I will update you again if there are any fixes.

zrdimetc, 2021-09-04:
[go-nmea](https://github.com/adrianmo/go-nmea) does not support NMEA TXT sentences. It's a warning message you can safely ignore. We use RMC sentence (_"Recommended Minimum Specific GPS/Transit data"_). If your u-block NEO-7 / VK-172 GPS dongle has a fix, then with gps option in xml config enabled. you should be getting GPS position reading after pressing F12. If `trackgpsshowlcd` option is also true, the position will be updated on talKKonnect LCD screen. Adjust traccar settings accordingly if you wish to use talKKonnect with [traccar server](https://www.traccar.org).

```
[  warn ] 2021/09/04 14:08:50 gps.go:283: Scanner Function Error  nmea: sentence prefix 'GPTXT' not supported
[  warn ] 2021/09/04 14:08:50 gps.go:283: Scanner Function Error  nmea: sentence prefix 'GPTXT' not supported
[  warn ] 2021/09/04 14:08:50 gps.go:283: Scanner Function Error  nmea: sentence prefix 'GPTXT' not supported
[  warn ] 2021/09/04 14:08:50 gps.go:283: Scanner Function Error  nmea: sentence prefix 'GPTXT' not supported
[  warn ] 2021/09/04 14:08:50 gps.go:283: Scanner Function Error  nmea: sentence prefix 'GPTXT' not supported
[  warn ] 2021/09/04 14:08:50 gps.go:283: Scanner Function Error  nmea: sentence prefix 'GPTXT' not supported
[  info ] 2021/09/04 14:08:50 gps.go:231: Date and Time from GPS:  04/09/21 12:08:50.0000
[  info ] 2021/09/04 14:08:50 gps.go:232: Date and Time from GPS Reformatted for Traccar:  2019-12-01 12:08:50
[  info ] 2021/09/04 14:08:50 gps.go:234: System Date with time.now Function: 2021-09-04
[  info ] 2021/09/04 14:08:50 gps.go:235: System Time with time.now function: 14:08:50
[  info ] 2021/09/04 14:08:50 gps.go:236: System Date & Time with time.now Function: 2021-09-04 14:08:50
[  info ] 2021/09/04 14:08:50 gps.go:238: GPS Date:  04/09/21
[  info ] 2021/09/04 14:08:50 gps.go:239: GPS Time:  12:08:50.0000
[  info ] 2021/09/04 14:08:50 gps.go:240: Validity:  A
[  info ] 2021/09/04 14:08:50 gps.go:241: Latitude Decimal:  44.780481666666674
[  info ] 2021/09/04 14:08:50 gps.go:242: Longitude Decimal:  20.515571666666666
[  info ] 2021/09/04 14:08:50 gps.go:243: Latitude DMS:  44° 46' 49.734000"
[  info ] 2021/09/04 14:08:50 gps.go:244: Longitude DMS:  20° 30' 56.058000"
[  info ] 2021/09/04 14:08:50 gps.go:245: Latitude GPS:  4446.8289
[  info ] 2021/09/04 14:08:50 gps.go:246: Longitude GPS:  2030.9343
[  info ] 2021/09/04 14:08:50 gps.go:247: Speed:  0.028
[  info ] 2021/09/04 14:08:50 gps.go:248: Course:  0
[  info ] 2021/09/04 14:08:50 gps.go:249: Variation:  0
[  info ] 2021/09/04 14:08:50 gps.go:250: Traccar Cmd Osmand:
[  info ] 2021/09/04 14:08:50 gps.go:251: Traccar $GPRMC Sentence for T55/OpenGTS: $GPRMC,120850.00,A,4446.82890,N,02030.93430,E,0.028,,040921,,,A*7F
[  info ] 2021/09/04 14:08:50 client.go:435: GPS Tracking is enabled
[ alert ] 2021/09/04 14:08:50 client.go:568: Client Is Already Registered
[  info ] 2021/09/04 14:09:20 gps.go:182: Showing GPS Info in LCD: Lat: 44.780481666666674 Long: 20.515571666666666
[ debug ] 2021/09/04 14:09:20 client.go:331: LCD Backlight Ticker Timed Out After 30 Seconds

``` 
[image]

zrdimetc, 2021-09-04:
If your talKKonnect restarts, after reading GPS position, check if ttyACM0 device is set in the config? Do you use LCD or OLED screen? If in RPi mode but screen is disconnected, talKKonnect could crash. Try the PC mode? Do you get GPS position with F12 pressed?

haocomm, 2021-09-04:
thank you for you information.

In my talkkonnect.xml is enable already.

**check if device name ttyACM0 is set in the config?** 

- device name is ttyACM0 and I use this 

<gps enabled="true">
	<port>/dev/ttyACM0</port>
	<baud>115200</baud>
	<txdata></txdata>
	<even>false</even>
	<odd>false</odd>
	<rs485>false</rs485>
	<rs485highduringsend>false</rs485highduringsend>
	<rs485highaftersend>false</rs485highaftersend>
	<stopbits>1</stopbits>
	<databits>8</databits>
	<chartimeout>100</chartimeout>
	<minread>0</minread>
	<rx>true</rx>
	<gpsinfoverbose>false</gpsinfoverbose>
</gps>

**Do you use LCD or OLED screen?** 
- I use OLED screen.

**Try the PC mode?**
- not test yet.

**Do you get GPS position in tty console screen with F12 pressed?**
- don't get position in tty console
  after press F12 
  **show error message:**  scanncer Function Error nmea: sentence prefix 'GPTXT' not supported
  then screen is terminating.

  see picture.
[http://mumble.haocomm.com/images/error-code-gps.png](url)

For GPS I check status
 tail -f /dev/ttyACM0 

$GPRMC,142414.00,A,1759.58990,N,09901.71514,E,0.205,,040921,,,A*7F
$GPVTG,,T,,M,0.205,N,0.379,K,A*29
$GPGGA,142414.00,1759.58990,N,09901.71514,E,1,06,1.40,562.3,M,-37.1,M,,*73
$GPGSA,A,3,16,26,22,32,27,31,,,,,,,2.91,1.40,2.56*09
$GPGSV,4,1,14,01,02,241,,03,42,291,,04,16,322,19,08,05,192,21*7E
$GPGSV,4,2,14,16,74,223,26,21,02,224,,22,48,261,31,26,65,023,17*77
$GPGSV,4,3,14,27,22,169,35,29,02,048,,31,31,037,16,32,28,115,23*7F
$GPGSV,4,4,14,40,36,252,,50,52,120,*79
$GPGLL,1759.58990,N,09901.71514,E,142414.00,A,A*6F
$GPTXT,01,01,02,u-blox ag - www.u-blox.com*50
$GPTXT,01,01,02,HW  UBX-G70xx   00070000 *77
$GPTXT,01,01,02,ROM CORE 1.00 (59842) Jun 27 2012 17:43:52*59
$GPTXT,01,01,02,PROTVER 14.00*1E
$GPTXT,01,01,02,ANTSUPERV=AC SD PDoS SR*20
$GPTXT,01,01,02,ANTSTATUS=OK*3B
$GPTXT,01,01,02,LLC FFFFFFFF-FFFFFFFD-FFFFFFFF-FFFFFFFF-FFFFFFF9*53
$GPRMC,142415.00,A,1759.58999,N,09901.71518,E,0.331,,040921,,,A*7D
$GPVTG,,T,,M,0.331,N,0.613,K,A*26
$GPGGA,142415.00,1759.58999,N,09901.71518,E,1,06,1.40,562.2,M,-37.1,M,,*76
$GPGSA,A,3,16,26,22,32,27,31,,,,,,,2.91,1.40,2.56*09
$GPGSV,4,1,14,01,02,241,,03,42,291,,04,16,322,19,08,05,192,20*7F
$GPGSV,4,2,14,16,74,223,26,21,02,224,,22,48,261,31,26,65,023,16*76
$GPGSV,4,3,14,27,22,169,34,29,02,048,19,31,31,037,17,32,28,115,23*77
$GPGSV,4,4,14,40,36,252,,50,52,120,*79
$GPGLL,1759.58999,N,09901.71518,E,142415.00,A,A*6B
$GPRMC,142416.00,A,1759.59022,N,09901.71530,E,0.392,,040921,,,A*75
$GPVTG,,T,,M,0.392,N,0.727,K,A*29
$GPGGA,142416.00,1759.59022,N,09901.71530,E,1,06,1.40,562.0,M,-37.1,M,,*75

haocomm, 2021-09-04:
Hi,  zrdimetc

Now it work.  thank you very much.
put this config in .xml
```
<gps enabled="true">
<port>/dev/ttyACM0</port>
<baud>115200</baud>
<txdata></txdata>
<even>false</even>
<odd>false</odd>
<rs485>false</rs485>
<rs485highduringsend>false</rs485highduringsend>
<rs485highaftersend>false</rs485highaftersend>
<stopbits>1</stopbits>
<databits>8</databits>
<chartimeout>100</chartimeout>
<minread>0</minread>
<rx>true</rx>
<gpsinfoverbose>false</gpsinfoverbose>
</gps>

<gpstrackingfunction enabled="true">
<traccarsendto>false</traccarsendto>
<traccarserverurl>http://10.8.0.1</traccarserverurl>
<traccarserverip>10.8.0.1</traccarserverip>
<traccarclientid>tk-demo-unit</traccarclientid>
<traccarreportfrequency>30</traccarreportfrequency>
<traccarproto>osmand</traccarproto>
<traccarserverfullurl></traccarserverfullurl>
<trackgpsshowlcd>true</trackgpsshowlcd>
<trackverbose>true</trackverbose>
</gpstrackingfunction>
```

zrdimetc, 2021-09-04:
We are glad to hear you made it work. You must have been missing `gpstrackingfunction` enabled in the config? It can be confusing, because of the history of gps functionality in talKKonnect. Originally, the gps function was working in relation to the "panic" button, When "gps" is enabled, its purpose was working with panic function to provide a position and secondary, printing the position to console screen for debugging. At some point, support for [traccar](https://github.com/traccar/traccar) was added by Suvir and `gpstrackingfunction` in the config is mainly about working with traccar. You can also enable printing the gps grid to LCD screen, which can be handy if traccar server connection suddenly drops. You can read position from the screen. In essence, you can mount your talKKonnect build in a car, track yourself and maintain a PTT communication from the road over a 3G network. In case of a need, you can send an alert by a button press. Alert message contains the gps grid and a predefined text message. It goes to Mumble channel and predefined email address. Audio can be automatically streamed from talKKonnect if `txlock` feature is enabled. There will be a ups functionality added to talKKonnect soon, to make powering the device on and off, from battery, in mobile and portable applications more easy.

talkkonnect, 2021-09-04:
Thank you Zoran for your support and for all your contributions for the traccar features you added to talkkonnect.

closed 2021-09-05 by talkkonnect

zrdimetc, 2022-02-07:
Update. GPS tracking function had been redesigned couple of weeks ago.  There are 3 tracking protocols supported: OsmAnd, T55 and OpenGTS. We use GPRMC, GPGGA and GPGSV NMEA sentences get useful information for tracking. GPS information can be shown in the display. If GPS or Traccar fails, audible alerts can be triggered.

GPS config did not changed much since the last time, 

```
<gps enabled="true">
<port>/dev/ttyACM0</port>
<baud>115200</baud>
<txdata/>
<even>false</even>
<odd>false</odd>
<rs485>false</rs485>
<rs485highduringsend>false</rs485highduringsend>
<rs485highaftersend>false</rs485highaftersend>
<stopbits>1</stopbits>
<databits>8</databits>
<chartimeout>100</chartimeout>
<minread>0</minread>
<rx>true</rx>
<gpsinfoverbose>true</gpsinfoverbose>
<gpsdiagsounds>true</gpsdiagsounds>
<gpsdisplayshow>true</gpsdisplayshow>
</gps>
```
New gps config settings:

`gpsinfoverbose` - will show more detailed GPS info in console screen
`gpsdiagsounds` - play diagnostic sounds
`gpsdisplayshow` - this shows GPS info in display (lat, long, speed)

Traccar config can look like this:

```
<traccar enabled="true">
<track>true</track>
<clientid>traccar_id</clientid>
<traccardisplayshow>true</traccardisplayshow>
<traccardiagsounds>true</traccardiagsounds>
<protocol name="osmand">
<osmand port="5055">
<serverurl>http://traccar_server</serverurl>
</osmand>
<t55 port="5005">
<serverip>traccar_server</serverip>
</t55>
<opengts port="5159">
<serverurl>http://traccar_server</serverurl>
</opengts>
</protocol>
</traccar>
```

New settings in Traccar config are:
`traccardisplayshow` - show traccar info in display
`traccardiagsounds` - play diagnostic sounds

You can enable GPS reading independently of Traccar, but obviosely GPS reading is a prerequisite for Traccar function working.
If GPS fails, no fix, or Traccar connection fails you will get errors warnings in console and display.
For setting a reporting rate for GPS tracking, at this stage of development just use `cron` and http api gps function. Program reporting function is under development.

Edit local `crontab` with `-e`.

This will report a position every minute

`* * * * * curl 'http://127.0.0.1:8080/?command=gpsposition'
`
If you need seconds, you can work around cron limitation like this,

```
* * * * * sleep 5; curl 'http://127.0.0.1:8080/?command=gpsposition'
* * * * * sleep 10; curl 'http://127.0.0.1:8080/?command=gpsposition'
* * * * * sleep 15; curl 'http://127.0.0.1:8080/?command=gpsposition'
```
...

Suvir's Readme and blog will be updated soon to reflect GPS functionality changes.