talkkonnect/talkkonnect #52: alert + error log

danybd, 2021-09-27:
Hello,
I setup a talkkonnect system but can not get rid of couple log which are showing up during startup. I copy/paste the entire log:

[  info ] 2021/09/27 20:26:26 Successfully Read file talkkonnect.xml
[  info ] 2021/09/27 20:26:26 Successfully loaded XML configuration file into memory
[  info ] 2021/09/27 20:26:26 Successfully Added Account talkkonnect-community to Index [0]
[  info ] 2021/09/27 20:26:26 client.go:123: Loglevel Set to Debug
[  info ] 2021/09/27 20:26:26 client.go:176: MQTT Server Subscription Disabled in Config
[  info ] 2021/09/27 20:26:26 client.go:229: Trying to Open File  /var/log/talkkonnect.log
[  info ] 2021/09/27 20:26:26 client.go:257: [1] Default Mumble Accounts Found in XML config
[  info ] 2021/09/27 20:26:26 client.go:267: Target Board Set as PC (gpio disabled)
[ debug ] 2021/09/27 20:26:26 client.go:299: Backlight Timer Disabled by Config
[  info ] 2021/09/27 20:26:26 talkkonnect.go:41: ┌────────────────────────────────────────────────────────────────┐
[  info ] 2021/09/27 20:26:26 talkkonnect.go:42: │  _        _ _    _                               _             │
[  info ] 2021/09/27 20:26:26 talkkonnect.go:43: │ | |_ __ _| | | _| | _____  _ __  _ __   ___  ___| |_           │
[  info ] 2021/09/27 20:26:26 talkkonnect.go:44: │ | __/ _` | | |/ / |/ / _ \| '_ \| '_ \ / _ \/ __|  __|         │
[  info ] 2021/09/27 20:26:26 talkkonnect.go:45: │ | || (_| | |   <|   < (_) | | | | | | |  __/ (__| |_           │
[  info ] 2021/09/27 20:26:26 talkkonnect.go:46: │  \__\__,_|_|_|\_\_|\_\___/|_| |_|_| |_|\___|\_ _|\__|          │
[  info ] 2021/09/27 20:26:26 talkkonnect.go:47: ├────────────────────────────────────────────────────────────────┤
[  info ] 2021/09/27 20:26:26 talkkonnect.go:48: │A Flexible Headless Mumble Transceiver/Gateway for RPi/PC/VM    │
[  info ] 2021/09/27 20:26:26 talkkonnect.go:49: ├────────────────────────────────────────────────────────────────┤
[  info ] 2021/09/27 20:26:26 talkkonnect.go:50: │Created By : Suvir Kumar  <suvir@talkkonnect.com>               │
[  info ] 2021/09/27 20:26:26 talkkonnect.go:51: ├────────────────────────────────────────────────────────────────┤
[  info ] 2021/09/27 20:26:26 talkkonnect.go:52: │Press the <Del> key for Menu or <Ctrl-c> to Quit talkkonnect    │
[  info ] 2021/09/27 20:26:26 talkkonnect.go:53: │Additional Modifications Released under MPL 2.0 License         │
[  info ] 2021/09/27 20:26:26 talkkonnect.go:54: │Blog at www.talkkonnect.com, source at github.com/talkkonnect   │
[  info ] 2021/09/27 20:26:26 talkkonnect.go:55: └────────────────────────────────────────────────────────────────┘
[  info ] 2021/09/27 20:26:26 talkkonnect.go:56: Talkkonnect Version 1.68.03 Released Sep 5 2021
[  info ] 2021/09/27 20:26:26 talkkonnect.go:57:
[ error ] 2021/09/27 20:26:26 client.go:307: Unable to Unmute  failed to execute "pactl set-sink-mute 0 0" (exit status 1)
[ debug ] 2021/09/27 20:26:26 onevent.go:61: Connected to talkkonnect-community Address [::1]:64738 on attempt 0 index [0]

[  info ] 2021/09/27 20:26:26 onevent.go:64: Welcome message:
[  info ] 2021/09/27 20:26:26 onevent.go:66:
[  info ] 2021/09/27 20:26:26 onevent.go:66:  Welcome to this server running Murmur.
[  info ] 2021/09/27 20:26:26 onevent.go:66:  Enjoy your stay!
[ alert ] 2021/09/27 20:26:26 client.go:427: Client Is Already Registered

What does mean "Client Is Already Registered"?
Where is coming the error "Unable to Unmute  failed to execute "pactl set-sink-mute 0 0" (exit status 1)" from?

zrdimetc, 2021-10-01:
Hello. You can choose to register your user name upon connection with Mumble server or not. Refer to [Mumble wiki](https://wiki.mumble.info/wiki/Murmurguide#Becoming_Administrator_and_Registering_a_User) for more info what does the client registration mean in Mumble and do you need it in your project or not? If you dont want or need to register user name, just set the `register `setting in `talkkonnect.xml` to `false`. There is a function that is muting the speaker in [talKKonnect](https://talkkonnect.com), when you are talking, with the intention of preventing feedback noise. It depends on [volume-go](https://github.com/talkkonnect/volume-go/blob/master/volume_linux.go). You can use [ALSA](https://www.alsa-project.org/wiki/Main_Page) - or some users prefer to use [Pulseaudio](https://www.freedesktop.org/wiki/Software/PulseAudio/). While some users use RPi `BCM` sound for output sound and USB sound device for input, other prefer USB sound card for both input and output. Depending on whether `BCM2835 `sound is disabled or not, Pulseaudio sink (output) device number will be 0 or 1... If you use Pulseaudio, you have to decide on your preferred method how to start it, run it system-wide and control it successfully. Refer to pulseaudio documentation for this. For `pactl` to work, you have to start `pulseaudio` and know the sink number of sink name. If your `BCM2835 `sound is still enabled and you have a USB sound card you use for input, then USB device will most likely be referenced as sink id 1. Disable BCM sound in `/boot/config.txt` with `dtparam=audio=off`, restart and try `pactl set-sink-mute 0 0` to unmute or 0 1 to mute. With BCM and USB sound to mute USB output, you need `pactl set-sink-mute 1 0` to unmute. 1 1 to mute. Use USB card for both sound input and output? Test [pactl](url) command from command line after you start pulseaudio? If you have some special requirements feel free to tweak `volume-go` or talKKonnect program files. We hope this helps?

zrdimetc, 2021-10-02:
If you run a headless [Raspberry Pi OS](https://www.raspberrypi.org/software) and dont need Pulseaudio, you can remove it by `apt-get remove pulseaudio` or `apt-get purge pulseaudio` and `apt-get autoremove`. Restart. Your pactl error will go away. If you run a desktop Raspberry Pi OS with pulseaudio package pre-installed, need pulseaudio and plan to use it with other programs, as well as with talKKonnect, you need to configure muting in Pulseaudio by setting device sink number in `talkkonnect.xml`
`<outputdevice>1</outputdevice>`
This will be the case if you have BCM sound enabled and it shows as "0" sink when you run `pactl list sinks` or `pactl list sinks short`
```
pactl list sinks short
0       alsa_output.platform-bcm2835_audio.analog-stereo        module-alsa-card.c      s16le 2ch 48000Hz       SUSPENDED
1       alsa_output.usb-0d8c_C-Media_USB_Headphone_Set-00.analog-stereo module-alsa-card.c      s16le 2ch 48000Hz       RUNNING
```
As you can see C-Media CM108 sound card is identified as sink 1? However, if you followed advice to disable BCM sound in `/boot/config.txt` then the BCM sound won't load and your USB sound card will be the sole sink "0"
```
pactl list sinks short
0       alsa_output.usb-0d8c_C-Media_USB_Headphone_Set-00.analog-stereo module-alsa-card.c      s16le 2ch 48000Hz       RUNNING
```
To start Pulseaudio... `pulseaudio --start`, to stop `pulseaudio -k`. There are additional considerations, if running Pulseaudio as root or service. `pactl` wont work unless you start `pulseaudio`. A user must be a member of `pulse-audio` group. Very good Wiki on Pulseaudio you can find [here](https://wiki.archlinux.org/title/PulseAudio). If you use ALSA for sound (Pulseaudio removed or not installed), then `outputdevice `name will be the `ALSA `device name. Name will typically be `Headphone`, `Speaker`... depending on how you configured `ALSA `sound. You can check the output device name through `alsamixer`. If you make local `.asoundrc`, input and output device names can be set there.

Have fun with talKKonnect.

talkkonnect, 2021-10-07:
Since There is no further updates or questions for more than a week, I will close this ticket. Again Thank you Zoran for supporting this issue.