talkkonnect/talkkonnect #55: Autoprovisioning - Can't Save File

ChZeman, 2021-12-28:
I set up autoprovisioning but it can't save the file. It **appears** that it's attempting to insert the path twice. If I leave <savefilepath> empty, it complains that options aren't set.

`<autoprovisioning enabled="true">
        <tkid>LunchBox-1</tkid>
        <url>http://server.com/talkkonnect</url>
        <savefilepath>/home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect</savefilepath>
        <savefilename>talkkonnect.xml</savefilename>
      </autoprovisioning>`

`[  info ] 2021/12/28 15:55:24 Contacting http Provisioning Server Pls Wait
[  info ] 2021/12/28 15:55:24 Trying to Autoprovision with URL:  http://server.com/talkkonnect/LunchBox-1.xml
[ debug ] 2021/12/28 15:55:24 HTTP Provisioning Server Responded With Status 200 OK
`

`Error from AutoProvisioning Module error: DownloadFile Module Returned an Error: "error: Cannot Create File Error: \"open /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect//home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/: no such file or directory\""`

Thank you,
Chris

talkkonnect, 2021-12-31:
Bug Fixed thank you for the testing and bug report Chris.

closed 2021-12-31 by talkkonnect

ChZeman, 2021-12-31:
Thank you, Suvir! I'd like to add that my fix renders the <savefilename> directive in talkkonnect.xml useless, but that probably doesn't matter unless someone wants to save it as a different filename for some reason.