talkkonnect/talkkonnect #57: How to transmit "aplay Systemsound"  as default input source

p1337, 2022-02-06:
Hallo there, 

I am running a raspi 4 with a mumble server and a talkkonnect client.
I have edited the client.go -> so that the client ist transmitting after connecting to the server.

What I want to do: is stream an "APLAY Systemsound Stream" to the Mumble Server, because I am running a rtl_sdr (software defined radio) that plays directly to aplay) 

`rtl_fm -f 87175000 -s 44100 -g 9 -l 0 - | aplay -t raw -r 44100 -c 2 -f S16_LE`

here comes the part, where I am not shure, if I had done it right.

I have activated the alsa loopback modules
I have uninstalled pulse audio, so the loopback is the default audio source ( I think, so )

Is there a best practice solution ?

Greetings from Stuttgart / germany
Robert

talkkonnect, 2022-02-06:
Never done this but have a look here
https://stackoverflow.com/questions/43480867/linux-pipe-audio-file-to-microphone-input

p1337, 2022-02-06:
Okay, time to fiddle it out - was to long. 
I took an USB-Soundcard - as "line-in" - and the soundoutput form the raspi built-in jack  and bridged them with an audio cable. 
Not the best way to solve this problem, I guess, but it worked within 2 minutes after a little bit of tweaking the settings.

talkkonnect, 2022-02-06:
Ok this use case was not ever imagined but do let us know if you find a clever hack using purely software. Since we are using pure alsa and not pulse audio I don’t know if a program like jack would work or not.

closed 2022-02-06 by talkkonnect

zrdimetc, 2022-02-06:
You had a good approach to use `snd_aloop` Alsa loopback module, but custom ALSA config is not something
you can spin up in a minute. For your case, you need to design `.asoundrc` and use ALSA plugs, 
like `dmix `and `dsnoop`. You are looking at some ALSA reading, making notes and experimenting.

Unofficial ALSA Wiki is here: [https://alsa.opensrc.org](https://alsa.opensrc.org)
ALSA forums are better place to ask ALSA config questions, provided you have gone through 
a lot of reading about ALSA and its traps: [https://github.com/opensrc/alsa/issues](https://github.com/opensrc/alsa/issues)
ALSA wiki info about .asoundrc is here: [https://www.alsa-project.org/main/index.php/Asoundrc](https://www.alsa-project.org/main/index.php/Asoundrc)
Another good ALSA source: [https://wiki.archlinux.org/title/Advanced_Linux_Sound_Architecture](https://wiki.archlinux.org/title/Advanced_Linux_Sound_Architecture)

[Pulseaudio](https://www.freedesktop.org/wiki/Software/PulseAudio) was created to simplify Linux sound management, but Pulseaudio has its many traps too.
Most tools for managing pulseaudio and [jackd](https://jackaudio.org) are desktop apps.

1. You can use [pavucontrol](https://freedesktop.org/software/pulseaudio/pavucontrol) "monitor" option to send output to input.
2. You can use tools like [QjackCtl](https://qjackctl.sourceforge.io) and [Patchage](https://drobilla.net/software/patchage.html) with jackd and pulseaudio / pulseaudio module jack to manage pulseaudio "sinks" and "sources" and "patch" 
them together.
3. To route rtl sdr audio to Mumble channel you could have used an official Muble client with "continuous" TX mode and one of the above methods involving pulseaudio, jackd and QjackCtl.

zrdimetc, 2022-02-07:
@p1337. Robert, here is a workaround for you for streaming rtl sdr source over talKKonnect to Mumble network... 
Install `vlc` and `sox` packages on your Pi.
Run: `rtl_fm -g50 -f 96.9M -M wfm -s 180k -E deemp | sox -traw -r180k -es -b16 -c1 -V1 - -t flac - | cvlc - --sout '#standard{access=http,mux=ogg,dst=:8888/audio.ogg}'`
In `talkkonnect,xml` set your http stream source: `<sound event="stream" file="http://your_ip_address:8888" volume="50" blocking="true" enabled="true"/>`
Get more vlc information here: `https://wiki.videolan.org/Documentation:Documentation` 
and here: `https://docs.videolan.me/vlc-user/3.0/en/advanced/index.html`
If you run as pi user, vlc will run fine. If running as root you have to tune privileges or just edit vlc binary:
```
root@tk-demo-04:~ # cvlc
VLC is not supposed to be run as root. Sorry.
If you need to use real-time priorities and/or privileged TCP ports
you can use /usr/bin/vlc-wrapper (make sure it is Set-UID root and
cannot be run by non-trusted users first).
root@tk-demo-04:~ # sed -i 's/geteuid/getppid/' /usr/bin/vlc
root@tk-demo-04:~ # cvlc
VLC media player 3.0.12 Vetinari (revision 3.0.12-1-0-gd147bb5e7e)
```
This is vlc playing piped audio from rtl_fm and sox player
```
root@tk-demo-04:~ # ./rtlsdr.sh
rtl_fm: no process found
play: no process found
aplay: no process found
/root/rtlsdr.sh: no process found
VLC media player 3.0.12 Vetinari (revision 3.0.12-1-0-gd147bb5e7e)
Found 1 device(s):
  0:  Realtek, RTL2832UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
[0014e4f8] main interface error: no suitable interface module
[000c7258] main libvlc error: interface "globalhotkeys,none" initialization failed
[0014e4f8] dummy interface: using the dummy interface module...
[6eb03030] mux_ogg mux: Open
Detached kernel driver
Found Rafael Micro R820T tuner
Tuner gain set to 49.60 dB.
Tuned to 97170000 Hz.
Oversampling input by: 6x.
Oversampling output by: 1x.
Buffer size: 7.59ms
Exact sample rate is: 1080000.025749 Hz
Allocating 15 zero-copy buffers
Sampling at 1080000 S/s.
Output at 180000 Hz.
```

Slight disadvantage of this approach is that you wont be able to use the same Pi for sending stream and receiving stream! One Pi will pipe `rtl_fm->sox->vlc`  and some other talKKonnect client needs to start F11 or http api for source streaming