talkkonnect/talkkonnect #58: Rotary Encoder Menu (feature request)

BeJay8675309, 2022-02-08:
Hi Developers,

Is it possible to add a simple menu structure to the rotary encoder so I don't have to use my custom python API frontend anymore? 

For basic control of the radio in either 7seg display or OLED, it would be ideal for the default rotary encoder input to be Volume, and pressing the knob (GPIO 12 on a Pi for example) would change to Channel mode, and back to volume after a delay of x seconds. Also the OLED screen would need more refresh updates as it sometimes doesn't completely refresh when changing volumes etc currently and leaves parts of the screen not updated. Thoughts?

Thanks

talkkonnect, 2022-02-09:
Hi BeeJay
There is current functionality to press the so called rotary encoder button to change mode between mumble channels, volume, sa818 frequency and voice target selection. This all works. Check latest release.

 There is however no timeout to go back to default mode and there is no indication on oled or seven segment yet. More development is needed and it is already on the road map. 

As for the menu what items or features do you control from your python app, please share. So if we find anything interesting we can add it to our future development roadmap.

we are also planning a redesign of the oled screens to make talkkonnect more informative and attractive. Looking for people to send in ideas and best contribute to programming these features.

BeJay8675309, 2022-02-09:
At the moment I just use channels and volume. They are all I need in the car for quick access. The OLED display is way to small to read whilst driving in it's single line mode, so I was using the adafruit oled library in python to make the channels and volume bigger so I can read them in a bigger font. I also had a startup screen that shows if internet is available before launching TK. 

This was going to be my solution, but I couldn't find the API callback or even a MQTT event at the time, to show users online/status and channel names, and even volume steps etc that I could bring into python. It may have changed since I wriote that, however I would rather just use TK directly without using my python layer to interface to the screen.

At the moment I use python without the OLED support and call the HTTP api events to turn the volume up and down and let TK display the volume on the OLED. along with the rest of the data. I will try the latest version and see how I go.

def encodervalue(rotaryenc):
	global encvalue, level, enctimer, changevol, loopcounter, talkinglevel
	print("* New value: {}".format(rotaryenc))
	if rotaryenc < encvalue:
		level = level -1
		#used to interface the talkkonnect api
		response = requests.get('http://localhost:8080/?command=VolumeDown')
		print(response)
		response = requests.get('http://localhost:8080/?command=CurrentVolume')
		print(response)
		if level < 0:
			level = 0
		encvalue = rotaryenc

	elif rotaryenc > encvalue:
		level = level + 1
		#used to interface the talkkonnect api
		response = requests.get('http://localhost:8080/?command=VolumeUp')
		print(response)
		response = requests.get('http://localhost:8080/?command=CurrentVolume')
		print(response)
		if level > 20:
			level = 20
		encvalue = rotaryenc

talkkonnect, 2022-02-15:
Hi BeeJay,
the latest code now supports rotary encoder mode change for channels, volume and voice target selection on seven segment using max7219. This should be good enough for your CB radio implementation. In the following releases I will add status updates  to the OLED also.

suvir kumar

talkkonnect, 2022-02-18:
No More comments so I will close this ticket

closed 2022-02-18 by talkkonnect