talkkonnect/talkkonnect #62: Onion route PTT

necro-nemesis, 2022-05-14:
I came across your project as I was researching the feasibility of adding a Mumble client to my own project. Found out coincidentally that you also were a ham operator. Among other things I code a headless access point that routes clients over a relatively new onion network that supports both udp and tcp called Lokinet. I also develop packages for this network. Mumble clients run flawlessly over this network and can access private mumble servers on the network using the networks own dns addresses. I was looking at potentially adding Mumble onion routed comms to my access point but after seeing the work you've done it would seem more logical to add the network as an option to talkkonnect thereby adding private onion routed calls to be made using PTT.

For reference here's my repo for the Pi based headless access point. I support both Raspberry and Orange Pi typically installed a Armbian.
https://github.com/necro-nemesis/LabyrinthAP

talkkonnect, 2022-05-15:
Very Nice project! How can I help do let me know

necro-nemesis, 2022-05-15:
Not sure yet but adding the network shouldn't be a big issue since there's already deb packages for both Arm and x86_64 running Debian or Ubuntu based distros (Armbian, Raspberry Pi OS etc.). There's also RPM packages if the distro is Fedora, openSUSE or Centos but I assume since you started with Arm and use SBC's, Debian variants and deb packages are probably the way to integrate the network. Technically in theory running the daemon on the system in conjunction should give talkkonnect users access to the network if it's similar to configuring a Mumble server or client. I just use an unmodifed version of Mumble with the daemon running on my access point. 

I wrote this wiki and in it explains how to add the repository and install the deb package for the network daemon. https://wiki.oxen.io/wiki/Main_Page#Lokinet_Install_for_Debian_.2F_Ubuntu_.2F_Mint

Let me know if there's something that could be done to improve user access to the network.  Other than providing privacy through onion routing udp and tcp packets there's another aspect that is beneficial about it in that the ports it uses also means that it can work through most standard firewall rules without having to forward ports on routers. It also has it's own dns so static IPs are possible without having to pay for static address. The combination of static IP and the port allocation means a very small server could be run behind a SOHO router.

talkkonnect, 2022-05-17:
I think your project is very complete, correct me if I am wrong but it seems it can work with your router out of the box using the /etc/hosts file etc and the tor network with normal IP. I am not so familiar with TOR, so really don't know how to take advantage of the TOR network. I will have to do some reading.

necro-nemesis, 2022-05-17:
You don't need to add addresses to the host file. The network has it's own dns resolution and you can set a persistent address when setting up a server through an config entry in the .ini file on the server making the server address persisitent vs ephemeral. DNS lookup is on 127.3.2.1. Depending on the distro used the daemon could provide lookup address through the stub-resolver for distros like Fedora that uses systemd-resolved or the address can be added to resolveconf's head file for Debian or Ubuntu. The name of the Mumble resolver could be the deamon's generated one or optionally it can be mapped to a human readable name using the systems name service. i.e. "talkkonnect.loki" for example. Hosting is explained under the SNApp section. https://wiki.oxen.io/wiki/Main_Page#Hosting_SNApps

necro-nemesis, 2022-05-17:
https://docs.oxen.io/products-built-on-oxen/lokinet/guides/secure-mumble-server-over-lokinet

talkkonnect, 2022-05-22:
use this as a guide for lokinet with talkkonnect
https://docs.oxen.io/products-built-on-oxen/lokinet/guides/secure-mumble-server-over-lokinet

closed 2022-05-22 by talkkonnect