talkkonnect/talkkonnect #63: Voice over TCP

Tobias0110, 2022-05-27:
Hi all,
I was setting up QoS for a Raspberry PI 3 running talkkonnect and saw that packets are only transmitted and received via the TCP port.
I know that the classic Mumble Client switches voice from UDP to TCP if UDP is not available. Therefore, I testet the connection with a laptop and the classic Mumble Client connected at the same point in my network as the raspberry before. In this configuration UDP was used for voice.

We run talkkonnect over a series of WIFI point-to-point connections with limited bandwidth. Due to this, QoS would be needed. Of course I can set up QoS for TCP but Mumble in TCP mode requires more bandwidth. Moreover UDP packets that are correctly marked as VoIP packets are generally handled with higher priority by most modern network hardware. In my tests talkkonnect showed audio dropouts in situations where the classic client using UDP had no problems.

My question is now: Does talkkonnect support UDP for voice? When yes, where can I enable it? 

Thanks!

zrdimetc, 2022-05-27:
You are correct. [talKKonnect](https://github.com/talkkonnect/talkkonnect) being a fork of [talkiepi](https://github.com/dchote/talkiepi) based on [gumble](https://github.com/layeh/gumble) `[is only working over tcp](https://github.com/layeh/gumble/issues/46)`. udp support was [assessed](https://github.com/layeh/gumble/issues/38) by Mr. Tim Cooper, author of gumble and `[udp branch](https://github.com/layeh/gumble/tree/udp)` was created to demonstrate udp working and for testing. However, there was not enough justification for merging udp code with main gumble code. You could try to build your own version of talKKonnect and slip in the udp changes from Tim's [udp branch](https://github.com/layeh/gumble/tree/udp). Mumble over tcp does require just a little more bandwidth for overhead. Your comparing of talKKonnect (over tcp) against the official Mumble client (over udp) and noticing smoother audio in the official client with less dropouts is probably just showing that official client is using a relatively efficient adaptive jitter buffer from Speex, while talKKonnect is not, while your network is sufficiently lossy to spot a difference. gumble is using some basic buffering and talKKonnect has a few tweaks and improvements compared to the original talkiepi / gumble code. Have a look inside `[stream.go](https://github.com/talkkonnect/talkkonnect/blob/main/stream.go)`. In great majority of use cases and networks, talKKonnect audio quality is perfectly acceptable. Suvir, talKKonnect author, had spent many sleepless nights working to improve the audio quality. There are little or no bandwidth limitations today for majority of users. For anyone planning to use talKKonnect in extremely lossy networks and low bandwidth, perhaps a jitter buffer can be designed, but this can create issues with latency. You can also sacrifice audio quality to conserve bandwidth by encoding audio at 8 KHz instead of 48. There was an [attempt to add static jitter buffer](https://github.com/layeh/gumble/pull/34) in gumble and talkiepi. In most of my VoIP projects tcp transport works a little better because of error control in lossy networks from udp which is fast but less reliable. Forcing tcp comes with a price of little more latency. If the network is well maintained and reliable, there are no issues either way. But if packets are lost persistently at 1-2% or higher rate, then no transport selection or clever buffering techniques in the application will help conceal the fact they are gone and this can not be masked or compensated any more.

zrdimetc, 2022-05-27:
One more advantage of tcp transport with Mumble ... both TLS control channel and voice are encrypted with TLS_AES_256_GCM_SHA384 cipher. With udp, Mumble voice is using 128-bit OCB-AES128 encryption, while TLS control channel is 256-bit AES. I can't say one is safer from other, but some users may have a preference.

talkkonnect, 2022-05-28:
Summarized perfectly by Zoran. You will have to look at gumble library. Or you cab reduce the sampling at the sacrifice of interoperability with other clients and lesser audio quality.

closed 2022-06-01 by Tobias0110