talkkonnect/talkkonnect #64: Question: Mqtt tts message

off112a, 2022-06-18:
Hi, 

Love your talkkonnect.

I wonder if it would be possible to make a mqtt command to send a tts message? For example a message that says that dinner is ready.

talkkonnect, 2022-06-18:
As of current version tts is supported by httpapi only. I have not yet made mqtt support text to speech. Should. Not be too difficult to do it. You can copy the concept from httpapi.go to mqtt.go if you can program in golang.

off112a, 2022-06-18:
Ok, thank you for your answer. Unfortunately, I have no experience in golang.

talkkonnect, 2022-06-18:
Will using httpapi solve your problem?

off112a, 2022-06-18:
I'm using Home Assistant with mqtt so I haven't tried httpapi. But I think it can be done if I have the right command for tts.

talkkonnect, 2022-06-19:
http://192.168.10.3:8080/?command=ttsannouncement&ttsmessage=%22dinner%20is%20ready%22&ttslocalplay=true&ttsplayintostream=false&gpioenabled=false&gpioname=relay0&language=en

change the 192.168.1.3 to your talkkonnect local ip ths command will announce dinner is ready over the speaker
alternatively you can test from your browser first

hope this solution works for you

off112a, 2022-06-22:
Thank you. It works perfect. 🙂

Do you think you will implement this for mqtt in the future?

talkkonnect, 2022-06-22:
That was the original plan to make announcements and tts work with mqtt. However these days very busy with work so less time to develop talkkonnect. Maybe in the future let’s see.

closed 2022-06-22 by talkkonnect