talkkonnect/talkkonnect #65: issue with orange pi zero

haocomm, 2022-06-28:
Hi, talkkonnect
I have some issue with orange pi zero.  

enviroment : 
HW : orange pi zero
OS: Armbian 22.05 Bullseye

I would like to used orange pi zero and I try to Install talkkonnect via the script tkbuild-orangepi.sh . 
and I have some error message detail as below.

get "periph.io/x/periph/conn/physic": found meta tag vcs.metaImport{Prefix:"periph.io/x/periph", VCS:"git", RepoRoot:"https://github.com/google/periph"} at //periph.io/x/periph/conn/physic?go-get=1
get "periph.io/x/periph/conn/physic": verifying non-authoritative meta tag
periph.io/x/periph (download)
cannot find package "periph.io/x/periph/conn/physic" in any of:
        /usr/local/go/src/periph.io/x/periph/conn/physic (from $GOROOT)
        /home/talkkonnect/gocode/src/periph.io/x/periph/conn/physic (from $GOPATH)
get "periph.io/x/periph/conn/spi": found meta tag vcs.metaImport{Prefix:"periph.io/x/periph", VCS:"git", RepoRoot:"https://github.com/google/periph"} at //periph.io/x/periph/conn/spi?go-get=1
get "periph.io/x/periph/conn/spi": verifying non-authoritative meta tag
cannot find package "periph.io/x/periph/conn/spi" in any of:
        /usr/local/go/src/periph.io/x/periph/conn/spi (from $GOROOT)
        /home/talkkonnect/gocode/src/periph.io/x/periph/conn/spi (from $GOPATH)
get "periph.io/x/periph/conn/spi/spireg": found meta tag vcs.metaImport{Prefix:"periph.io/x/periph", VCS:"git", RepoRoot:"https://github.com/google/periph"} at //periph.io/x/periph/conn/spi/spireg?go-get=1
get "periph.io/x/periph/conn/spi/spireg": verifying non-authoritative meta tag
cannot find package "periph.io/x/periph/conn/spi/spireg" in any of:
        /usr/local/go/src/periph.io/x/periph/conn/spi/spireg (from $GOROOT)
        /home/talkkonnect/gocode/src/periph.io/x/periph/conn/spi/spireg (from $GOPATH)
get "periph.io/x/periph/devices/apa102": found meta tag vcs.metaImport{Prefix:"periph.io/x/periph", VCS:"git", RepoRoot:"https://github.com/google/periph"} at //periph.io/x/periph/devices/apa102?go-get=1
get "periph.io/x/periph/devices/apa102": verifying non-authoritative meta tag
cannot find package "periph.io/x/periph/devices/apa102" in any of:
        /usr/local/go/src/periph.io/x/periph/devices/apa102 (from $GOROOT)
        /home/talkkonnect/gocode/src/periph.io/x/periph/devices/apa102 (from $GOPATH)
get "periph.io/x/periph/host": found meta tag vcs.metaImport{Prefix:"periph.io/x/periph", VCS:"git", RepoRoot:"https://github.com/google/periph"} at //periph.io/x/periph/host?go-get=1
get "periph.io/x/periph/host": verifying non-authoritative meta tag
cannot find package "periph.io/x/periph/host" in any of:
        /usr/local/go/src/periph.io/x/periph/host (from $GOROOT)
        /home/talkkonnect/gocode/src/periph.io/x/periph/host (from $GOPATH)
        
Can you  recommend? 
Thank you.

zrdimetc, 2022-06-29:
The problem is that [periph.io](https://periph.io) library used for working with spi APA102 RGB leds in [ReSpeaker 2 Mic Array](https://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT) audio hat - had a big make over between 2020 and now. The old periph.io code was broken down to a number of [smaller git repositories](https://github.com/periph) with different paths. Old periph.io source code was removed. The newly supported [periph,io](https://github.com/periph) code is in github, not in periph,io repository any more. As a result [ledstrip.go](https://github.com/talkkonnect/talkkonnect/blob/main/ledstrip.go) needs some reworking for the new periph.io or an older version of periph.io will have to be forked in talkkonnect project and remain frozen, no longer supported by periph.io author, no updates. periph,io is an excellent project, but a small overkill for just driving spi RGB led. Perhaps a light-weight solution without too many dependencies for driving RGB led will be found at some point or the existing periph.io APA102 code can be trimmed down to just the essential, keeping it simple? Ideal spi RGB led library should support WS8211, WS8212B and other low cost led's, in addition to APA102 which is harder to source than WS821X. This could give talkkonnect builders more freedom to use a variety of RGB leds in any shape or form, not just APA102 from ReSpeaker.

zrdimetc, 2022-06-29:
As a temporary workaround, replace broken periph.io inside `/home/talkkonnect/gocode/src` folder with the attached old periph.io library. Delete broken periph,io, unzip attached file to src folder and recompile.
[periph.io.zip](https://github.com/talkkonnect/talkkonnect/files/9015082/periph.io.zip)

haocomm, 2022-06-30:
Thank you very much.    [zrdimetc](https://github.com/zrdimetc)

talkkonnect, 2022-07-03:
When I have time I will put this library into talkkonnect repository and freeze it (vendor it) so talkkonnect wont break when external changes to this perip.io are made.

talkkonnect, 2022-07-10:
I Have vendored periph.io library and put in talkkonnect repository now talkkonnect can build normally.

closed 2022-07-10 by talkkonnect