talkkonnect/talkkonnect #67: One way Audio  USb device error

rajeevkk32, 2022-09-14:
hi,

error while speaking from a client

Device: Raspberry PI 4
Cmedia USB Audio

Sep 14 17:57:14 poc-gw-7 kernel: [  399.505490] retire_capture_urb: 1969 callbacks suppressed
Sep 14 17:57:15 poc-gw-7 kernel: [  400.772625] usb 1-1.3: reset full-speed USB device number 11 using xhci_hcd
Sep 14 17:57:15 poc-gw-7 kernel: [  401.222548] usb 1-1.3: reset full-speed USB device number 11 using xhci_hcd
Sep 14 17:57:15 poc-gw-7 kernel: [  401.322714] usb 1-1.3: Device not responding to setup address.
Sep 14 17:57:16 poc-gw-7 kernel: [  401.552526] usb 1-1.3: Device not responding to setup address.
Sep 14 17:57:16 poc-gw-7 kernel: [  401.872368] usb 1-1.3: reset full-speed USB device number 11 using xhci_hcd
Sep 14 17:57:16 poc-gw-7 kernel: [  401.872552] usb 1-1.3: Device not responding to setup address.
Sep 14 17:57:16 poc-gw-7 kernel: [  402.092588] usb 1-1.3: Device not responding to setup address.
Sep 14 17:57:17 poc-gw-7 kernel: [  402.422374] usb 1-1.3: reset full-speed USB device number 11 using xhci_hcd
Sep 14 17:57:17 poc-gw-7 kernel: [  402.422562] usb 1-1.3: Device not responding to setup address.
Sep 14 17:57:17 poc-gw-7 kernel: [  402.642552] usb 1-1.3: Device not responding to setup address.
Sep 14 17:57:17 poc-gw-7 kernel: [  402.863385] usb 1-1.3: USB disconnect, device number 11
Sep 14 17:57:17 poc-gw-7 kernel: [  403.152393] usb 1-1.3: new full-speed USB device number 12 using xhci_hcd
Sep 14 17:57:17 poc-gw-7 kernel: [  403.252731] usb 1-1.3: Device not responding to setup address.
Sep 14 17:57:18 poc-gw-7 kernel: [  403.602397] usb 1-1.3: new full-speed USB device number 13 using xhci_hcd
Sep 14 17:57:18 poc-gw-7 kernel: [  403.922745] usb 1-1.3: Device not responding to setup address.
Sep 14 17:57:18 poc-gw-7 kernel: [  404.142573] usb 1-1.3: Device not responding to setup address.
Sep 14 17:57:18 poc-gw-7 kernel: [  404.362749] usb 1-1-port3: attempt power cycle
Sep 14 17:57:21 poc-gw-7 kernel: [  406.432426] usb 1-1.3: new full-speed USB device number 15 using xhci_hcd
Sep 14 17:57:21 poc-gw-7 kernel: [  406.579019] usb 1-1.3: New USB device found, idVendor=08bb, idProduct=2902, bcdDevice= 1.00
Sep 14 17:57:21 poc-gw-7 kernel: [  406.579032] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Sep 14 17:57:21 poc-gw-7 kernel: [  406.579039] usb 1-1.3: Product: USB PnP Sound Device
Sep 14 17:57:21 poc-gw-7 kernel: [  406.579045] usb 1-1.3: Manufacturer: C-Media Electronics Inc.      
Sep 14 17:57:21 poc-gw-7 kernel: [  406.656545] input: C-Media Electronics Inc.       USB PnP Sound Device as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.3/0003:08BB:2902.0006/input/input7
Sep 14 17:57:21 poc-gw-7 kernel: [  406.722828] hid-generic 0003:08BB:2902.0006: input,hidraw0: USB HID v1.00 Device [C-Media Electronics Inc.       USB PnP Sound Device] on usb-0000:01:00.0-1.3/input3
Sep 14 17:57:21 poc-gw-7 mtp-probe: checking bus 1, device 15: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Sep 14 17:57:21 poc-gw-7 mtp-probe: bus: 1, device: 15 was not an MTP device
Sep 14 17:57:21 poc-gw-7 mtp-probe: checking bus 1, device 15: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Sep 14 17:57:21 poc-gw-7 mtp-probe: bus: 1, device: 15 was not an MTP device

talkkonnect, 2022-09-15:
looks like a driver or linux issue. Don't think it has anything to do with talkkonnect. Did you use the image to install? Which Image? Did you install from scratch? please provide more details

talkkonnect, 2022-09-20:
Since no answer or response I will close this ticket

closed 2022-09-20 by talkkonnect