talkkonnect/talkkonnect #69: Install in OrangePi Zero2 | Ubuntu

litmus-paper, 2022-09-18:
Thank you for this great product.

I have installed Talkkonnect on OrangePi Zero2 running Ubuntu 22.04 - Jammy server. This ubuntu version is 64bit OS. During the process, I've run into an issue with the [periph/x/periph/host/host_arm64.go ](https://github.com/talkkonnect/periph/blob/main/x/periph/host/host_arm64.go) file. This file is missing the "github.com" prefix on the imports.
As a workaround, I have edited the file $GOPATH/src/github.com/talkkonnect/periph/x/periph/host/host_arm64.go
```
package host

import (
        // Make sure CPU and board drivers are registered.
        _ "github.com/talkkonnect/periph/x/periph/host/allwinner"
        _ "github.com/talkkonnect/periph/x/periph/host/bcm283x"
        _ "github.com/talkkonnect/periph/x/periph/host/pine64"
        _ "github.com/talkkonnect/periph/x/periph/host/rpi"
)
```

In addition, one needs to install pkg-config manually by running" 
`apt install pkg-config`

talkkonnect, 2022-09-20:
thank you for this feedback I am sure it will help others

talkkonnect, 2022-09-20:
I have made some comments in the host_arm64.go file and also added the pkg-config to the apt get script for the intstallers. Again thank you for your kind feedback and solution.

closed 2022-09-20 by talkkonnect