talkkonnect/talkkonnect #71: Disabling "on-board" PTT-Button when external PTT is plugged in

ItsGiantCookie, 2022-12-22:
This might be a novice question, since I haven't had much time to tinker around with the project yet but I would like to use talkkonnect on my Orange Pi with a NATO 6-Pin Connector attached to it, would it be possible to configure my build in a way to disable the PTT-Button on the case of my PI when the external Mic and PTT is connected to the U229 connector?

And if it's possible how would I go about changing the config to achieve this?

talkkonnect, 2022-12-25:
What Kind of hardware are you using that gives you this PTT Button on the case? I mean are you using GPIO directly or using a respeaker hat? You can have 2 different GPIO for PTT, so there would be no real reason to disable the other PTT on the other GPIO, I really need to understand your use case to make some sort of meaningful recommendation.

talkkonnect, 2022-12-30:
1. First of all try to log in as root with password talkkonnect using ssh
2. nano /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/talkkonnect.xml
3. fix the file below for example you want to add another gpio copy and duplicate the line below and set another pinno for the 2nd ptt , the example below will set ptt key on pinno 4 if you want to duplicate on pinno 5 for the same function to have 2 gpio one for the ptt button on the panel and one for ptt on the microphone you can do it by duplicating the line with a different pinno.

<pin direction="output" device="led/relay"     name="transmit" pinno="4" type="gpio" chipid="0" inverted="false" enabled="true"/>

talkkonnect, 2023-01-02:
Since there is no further questions or development let me close this ticket

closed 2023-01-02 by talkkonnect