talkkonnect/talkkonnect #72: Docker images?

sgofferj, 2022-12-26:
I would like to suggest the project provides docker images for different platforms. E.g. a Raspberry Pi 4 should be powerful enough to run other things besides talkkonnect and docker would make the software deployment and management much easier.

talkkonnect, 2022-12-27:
Hi Sgofferj,
I do not provide docker images at the moment because I have yet to learn docker. However if there are some volunteers to create such docker images I would be happy to host them for others. The images I have provided are also  pretty old need to update them when I get time.

sgofferj, 2022-12-27:
I have started learning docker a while ago and I am just making my first images for some of my projects. Nothing in go, though. As soon as I get my hands on a new RasPi, I'll start playing with Talkkonnect and maybe I can figure something out then.

talkkonnect, 2022-12-28:
Ok If you do figure something out do let me know so I can learn from you and the whole talkkonnect community can benefit from this improvement. We perhaps then can share a docker image with some short instructions of how to get it going on raspberry pi or other similar sbc.

closed 2022-12-28 by talkkonnect