talkkonnect/talkkonnect #74: Build script for x86_64 architecture?

whyronimus, 2023-02-27:
Hello,
I'm trying to build talkkonnect on a Debian (bullseye) machine. 
Is there a build script like the tkbuild-64bit.sh for this? If not, could it be added?

(if I'm trying something impossible, please let me know).

talkkonnect, 2023-02-28:
Hi,
compiling for x86 on some machines gives problems with the opus library, on other machines it just compiles without any problems. So there is no exact science there, or at least I do not have enough variants of the x86 machines to find the solution. You can modify the existing 64 bit script to get the correct golang version for your processor and everything else should just work. No need to change anything else in the script except the golang compiler.

whyronimus, 2023-02-28:
I changed the line:

`cd /usr/local
wget https://go.dev/dl/go1.19.linux-arm64.tar.gz
tar -zxvf go1.19.linux-arm64.tar.gz`

to

`cd /usr/local
wget https://go.dev/dl/go1.20.1.linux-amd64.tar.gz
tar -zxvf go1.20.1.linux-amd64.tar.gz`

Compilation does not show any errors, but still I get no binary files in /home/talkkonnect/bin.

Is there anything I am doing wrong? Let me know if I can provide any useful output.

Many TIA!

talkkonnect, 2023-03-01:
Can you share the whole script? So I can test on my side just to see. Any logs would be helpful as I don't know the problems that happened on your side without any logs or at least send me the real script you used. Also some info on your os and pc arch.

talkkonnect, 2023-03-04:
I also cant get gopus and cgo on golang to play nice with each other, some error about a 32 bit and 64 bin variable size incompatibility on the x86 arch. If anyone who is expert in C and C-GO please help.

talkkonnect, 2023-03-07:
Since I have no solution for this allow me to close this ticket

closed 2023-03-07 by talkkonnect