talkkonnect/talkkonnect #76: .img update Pi3,Pizero2w

f59465-Clem, 2023-02-28:
Hi to all,
I arrived at Talkkonnect to make a Voip-RF gateway
(Pizero2w + Eth/USB Hat and RF radio GM-360) according to a lot of positive reviews.

I installed an image of Pi3 (USB Sound Card) facing problems on the Oled 1.3 "display and the reverse state of the outputs, new function missing, Follow my advice to update, unfortunately without success, I tried (from Bash Script) obsolete.

I tried for several hours (from Source code by hand)
I'm probably not successful due to my amateur level.

And you it possible to have a new image which adds the new function, patch for Pi3, PiZero2W, 64bits, USB Sound Card ?

Cordially
from France

talkkonnect, 2023-03-01:
OK will do on weekend probably Sunday, thank you for your patients. Very busy with work now, very little time to play with talkkonnect.

f59465-Clem, 2023-03-01:
Hello, 
thank you for this good news,
 have a nice day.

talkkonnect, 2023-03-03:
The New Image for 64Bit Raspberry Pi 3,3b+,3A and Zero2W with USB sound card and OLED working released 03/03/2023 is available at the link https://drive.google.com/file/d/1w_900dLt3RNNhOoFNvW9RfzYXrP1iaRZ/view?usp=sharing

closed 2023-03-03 by talkkonnect