talkkonnect/talkkonnect #77: Error when loading

lucematt, 2023-03-10:
I'm trying to get your application to work on a pi 3B with the latest raspberry pi OS 64-bit installed on it.  I'm using the latest golang as well with 1.20.1 arm64.  

After launch the screen comes up with text, but then it exists so fast I'm unable to read what it says.  I'm just left with this error message.  No text in /var/log/talkkonnect.log.  Looking for any suggestions on how to troubleshoot this.

` /home/talkkonnect/bin/talkkonnect
panic: runtime error: index out of range [0] with length 0

goroutine 1 [running]:
github.com/talkkonnect/talkkonnect.(*Talkkonnect).ClientStart(0x4000390000)
        /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/client.go:540 +0x25b8
github.com/talkkonnect/talkkonnect.Init({0xa2fbb6, 0x4f}, {0x9fde7b, 0x1})
        /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/client.go:235 +0xf00
main.main()
        /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/cmd/talkkonnect/main.go:78 +0x2c4
`

lucematt, 2023-03-10:
Well I was about to grab the text by just redirecting it to a log.  I'm still not sure what I'm looking at yet as far as the problem goes.

[  info ] 2023/03/10 16:37:48 Successfully Read file talkkonnect.xml
[  info ] 2023/03/10 16:37:48 Starting XML Configuration Sanity and Logical Checks
[  warn ] 2023/03/10 16:37:48 Config Error [Section Hardware] VoiceActivityTimersecs < 200 setting to 200
[  warn ] 2023/03/10 16:37:48 Non-Critical Errors Found In talkkonnect.xml config file please fix errors or talkkonnect may not behave as expected
[  info ] 2023/03/10 16:37:48 Successfully loaded XML configuration file into memory
[  info ] 2023/03/10 16:37:48 Default Loglevel unset in XML config automatically loglevel to Info
[  info ] 2023/03/10 16:37:48 MQTT Server Subscription Disabled in Config
[  info ] 2023/03/10 16:37:48 Connecting to Server internal:64738 Identified As talkkonnect-community-server With Username talkkonnect-b827ebd6e651
[  info ] 2023/03/10 16:37:48 Trying to Open File  /var/log/talkkonnect.log
[  info ] 2023/03/10 16:37:48 [1] Default Mumble Accounts Found in XML config
[  info ] 2023/03/10 16:37:48 Target Board Set as RPI (gpio enabled)
[  info ] 2023/03/10 16:37:48 ┌────────────────────────────────────────────────────────────────┐
[  info ] 2023/03/10 16:37:48 │  _        _ _    _                               _             │
[  info ] 2023/03/10 16:37:48 │ | |_ __ _| | | _| | _____  _ __  _ __   ___  ___| |_           │
[  info ] 2023/03/10 16:37:48 │ | __/ _` | | |/ / |/ / _ \| '_ \| '_ \ / _ \/ __|  __|         │
[  info ] 2023/03/10 16:37:48 │ | || (_| | |   <|   < (_) | | | | | | |  __/ (__| |_           │
[  info ] 2023/03/10 16:37:48 │  \__\__,_|_|_|\_\_|\_\___/|_| |_|_| |_|\___|\_ _|\__|          │
[  info ] 2023/03/10 16:37:48 ├────────────────────────────────────────────────────────────────┤
[  info ] 2023/03/10 16:37:48 │A Flexible Headless Mumble Transceiver/Gateway for RPi/PC/VM    │
[  info ] 2023/03/10 16:37:48 ├────────────────────────────────────────────────────────────────┤
[  info ] 2023/03/10 16:37:48 │Created By : Suvir Kumar  <suvir@talkkonnect.com>               │
[  info ] 2023/03/10 16:37:48 ├────────────────────────────────────────────────────────────────┤
[  info ] 2023/03/10 16:37:48 │Press the <Del> key for Menu or <Ctrl-c> to Quit talkkonnect    │
[  info ] 2023/03/10 16:37:48 │Additional Modifications Released under MPL 2.0 License         │
[  info ] 2023/03/10 16:37:48 │Blog at www.talkkonnect.com, source at github.com/talkkonnect   │
[  info ] 2023/03/10 16:37:48 └────────────────────────────────────────────────────────────────┘
[  info ] 2023/03/10 16:37:48 Software Ver 2.19.01 Rel Mar 3  2023 (Latest Release)
[  info ] 2023/03/10 16:37:48 Hardware Detected As Raspberry Pi 3 Model B Rev 1.2
[ error ] 2023/03/10 16:37:48 Unable to Unmute  failed to execute "amixer set PCM unmute" (exit status 1)
[  info ] 2023/03/10 16:37:48 Connected to talkkonnect-community-server Address 10.10.20.200:64738 on attempt 0 index [0]
[  info ] 2023/03/10 16:37:48  Welcome to this server, running the official Mumble Docker image.
[  info ] 2023/03/10 16:37:48  Enjoy your stay!
[  info ] 2023/03/10 16:37:48 Current Channel  Root  has ( 1 ) participants
[  info ] 2023/03/10 16:37:48 1. User "talkkonnect-b827ebd6e651" is online. []
[  info ] 2023/03/10 16:37:48 Channel  Root  has no other participants
[  info ] 2023/03/10 16:37:49 Client Is Already Registered
panic: runtime error: index out of range [0] with length 0

goroutine 1 [running]:
github.com/talkkonnect/talkkonnect.(*Talkkonnect).ClientStart(0x4000320000)
        /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/client.go:540 +0x25b8
github.com/talkkonnect/talkkonnect.Init({0xa2fbb6, 0x4f}, {0x9fde7b, 0x1})
        /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/client.go:235 +0xf00
main.main()
        /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/cmd/talkkonnect/main.go:78 +0x2c4

lucematt, 2023-03-10:
``

lucematt, 2023-03-10:
Didn't mean to close it

talkkonnect, 2023-03-14:
You have a wrong setting in XML config. You must have at least 1 server enabled to true. Please check for logical errors in the XML file. Otherwise use the sample file provided and start modifying line by line.

talkkonnect, 2023-03-14:
Please try with this config first https://raw.githubusercontent.com/talkkonnect/talkkonnect/main/sample-configs/talkkonnect-version2-usb-gpio-example.xml

lucematt, 2023-03-14:
I'm not sure what xml setting was missing, but that other xml example you suggested does allow me to stay connected to the server.  

I was using https://raw.githubusercontent.com/talkkonnect/talkkonnect/main/sample-configs/talkkonnect-version1-raspberrypi-OLED-i2c.xml with just the gps and oled both set to false.  I reverted back to a fresh copy of that version1 file and turned gps and old to false again to test if I had messed something else up in the file originally and the results are the same.  It fails with a runtime error.

Propeller-v8, 2023-03-14:
If you used one of the images provided here on github, these images are images version2 of talkkonnect and therefor you have to use version2 config XML file.
Greets Andre

lucematt, 2023-03-14:
I did not use the images, manual install method.

talkkonnect, 2023-03-15:
Since it seems lucematt was using incomparable version 1 of config for version 2 of software, so i will close this ticket.
thanks for all your feedback and support from other users.

closed 2023-03-15 by talkkonnect