talkkonnect/talkkonnect #78: Raspberry Pi Zero W with ReSpeakerHat image is not accessible

ahgsql, 2023-03-13:
https://drive.google.com/file/d/15EJ84oFKAFWkdF4191Xi_kPz3E8MC1J0/view?usp=sharing

this link is from version1 branch, but is not accessible anymore. Need update or if someone has it, please send me a working link

talkkonnect, 2023-03-14:
Removed the link for raspberry pi zero version 1 (old version) because of many complaints of audio quality, The cpu of the pi zero w (old version) is simply not powerful enough. As a result it is removed. Please use Raspberry Pi Zero 2W instead.

talkkonnect, 2023-03-14:
I should have removed all references to the pi zero version 1, can you please advise where you found this dead link., so I can remove it as to help others who may try to download it in the future.

talkkonnect, 2023-03-15:
Since no activity or further development, allow me to close this ticket.

closed 2023-03-15 by talkkonnect

ahgsql, 2023-05-06:
Sorry for delayed reply, need that link still, if you have please share it with me as i bought w1 only for this project :)

talkkonnect, 2023-05-06:
Really sorry no more version 1 anymore. You can download the version 2 and then fix the code and recompile to get it to work. But the performance is horrible, not recommended.